edwardkort/WWIDesigner #22: Save As with validation errors changes view name without saving

bhp1, 2015-05-26:
If an instrument or tuning has validation errors, using "Save As" with a new name will change the name on the view tab, even though the data wasn't saved.  When the user closes the view (or the whole application), they get a "file save" warning, even if "warn on dirty close" is not set.

bhp1, 2015-05-26:
This behaviour is buried in the application framework.  Overriding saveData in WIDesigner class, or catching dataModelSaving in the DataModelAdapter, are not sufficient to avoid the change of name.

bhp1, 2015-05-26:
Removal of dataModel.isDirty() call from dataModelClosing appears to get rid of the "file save" warning whether or not "warn on dirty close" is set.  Under the circumstances of this issue, the call returned false, so the setDirty(false) call was skipped.
"File save" warning fixed for release 1.0.5.
(This does not address the change of name in the view tab.)

edwardkort, 2015-05-28:
There is a bug in JDAF. When I get a fix or workaround, I will fix this.

edwardkort, 2017-02-01:
As of 1/31/2017 JideSoft said that this bug will not be fixed: it requires too big a change in the JDAF architecture. This will remain a known issue unless we want to build our own validation framework. However, since we do very little XML hand editing any more, this issue IMHO is not a very high priority.