edwardkort/WWIDesigner #24: Mac OS 10.10.3

stanleyloftness, 2015-05-27:
The program operates in the mac environment, but has some problems

No toolbar
When you optimize a flute the new untitled flute does not show up under instruments.

edwardkort, 2015-05-27:
Thanks, Stanley. Did the program write any error/warning messages to the Console?

stanleyloftness, 2015-05-27:
No error messages.
Stan

Date: Wed, 27 May 2015 14:22:17 -0700

edwardkort, 2015-05-27:
Stan,
We are looking into it. One of the developers has a Mac to try it on.

edwardkort, 2015-05-28:
We have confirmed Stan's observations and identified 5 concerns:
1. No toolbar
2. No additional menu items in File menu
3. Newly created files do not populate the Study view
4. The sizing of the fingering widget isn't quite right
5. The NAF instrument view has overlapping cells.

We are working to resolve all of the above. Windows user should note that these are not issues on that platform.

edwardkort, 2015-05-28:
We have put a number of good-practices fixes in the code for the v1.0.5 release. However, none seem to effect any change. So for now, the Mac will be considered an unsupported platform.

bhp1, 2015-05-28:
Regarding toolbar: The JDAF documentation says: "application toolbars are *disabled by
default on Mac OS X* because they are application dependent and non-standard.  Use the pre-run option setToolBars(true) to turn on toolbars for OS X. On Leopard, JDAF will prepare the toolbar to work as a 'unified toolbar', but to complete this effect you must call the OSXProperties.setBackgroundEffect(OSXProperties.UNIFIED_TOOLBAR) prior to instantiating the GUIApplication. You may then use OSXProperites when building your UI to create compliant toolbar buttons." (Emphasis mine.)

I have not found any setToolBars(boolean) method in JDAF.

stanleyloftness, 2015-05-29:
I am not computer literate enough to understand how to do this fix.
Stan

Date: Thu, 28 May 2015 16:59:09 -0700

bhp1, 2015-05-29:
Sorry for the misunderstanding, Stanley.  My post wasn't directed at you.  It was technical information intended for the other developers.

edwardkort, 2015-06-01:
Fixed for v1.0.5 release. Still at issue, but not being actively addressed:
1. The Look and Feel for Macs has been set to cross-platform in order to display cell lines in tables, make the Hole Group Creation dialog usable, and the warning/information dialogs readable
2. The standard toolbar components (Open, Save, etc.) are not displayed, only the custom toolbar. Since toolbars are not part of the standard Mac interface, this issue is not considered important
3. The naming of new, unsaved files (from an optimization run or from Create default/blank constraints) is a bit inconsistent, starting with "Untitled" and then skipping to "Untitled5" or "Untitled3" respectively.

closed 2015-06-02 by edwardkort