edwardkort/WWIDesigner #32: no options shown in edit menu

rabfulton, 2015-06-11:
WIDesigner version: 1.0.6
java version: openjdk version "1.7.0_76"
OS: Freebsd 10.1

Program starts fine, but under the edit menu I only see cut, copy, paste, undo, redo, select all. There is no entry for options! Everything else appears to work.

What am I missing?

edwardkort, 2015-06-11:
The UI framework puts the Options menu item in different places, depending on the operating system. Under Windows, it is called Options and is under the Edit menu. On the Mac, it is called Preferences and is under the global WIDesigner menu. I don’t know where it is on FreeBSD, but look around in the places you would normally find a Preferences menu item.

BTW, you must have grabbed v1.0.6 during the brief time before I withdrew it. The real v1.0.6 will be published late today or tomorrow. You likely won’t have problems with the version you have, but the new optimizers I added don’t converge as well as the to-be published version.

I’d be interested in a few screenshots of how WIDesigner presents under FreeBSD.

rabfulton, 2015-06-11:
Its not in any of the menus, I have added some screenshots for you.
Anything I could try to get more information? I tried 1.0.5, looks the same.

[image]
[image]

edwardkort, 2015-06-11:
Thanks for the screenshots. Looks like the program is functional, other than the missing Options. What windowing interface are you using? Gnome or ...?

In other applications, where do you "normally" find the "Preferences ..." menu item? And what is it called?

I will look at the documentation for the UI library we are using, and see if there is anything unique to FreeBSD and whatever windowing interface you are using. But that won't be until this evening.

BTW, there is a 1.8.0_45 java version available. That might be the problem, but ...

Thanks for your help in tracking this down. I have no access to any Linux systems, so these L&F issues are hard to resolve.

rabfulton, 2015-06-11:
I use Openbox as a window manager. I would normally expect preferences under edit in most apps. I thought java UI would be the same on all systems. I think apple makes there own java release, hence the differences there?

In another java app called "BlueJ" it is tools->preferences.
How exactly does the UI framework decide? Which ui Framework are you referencing? swing? You could try linux or freebsd under a virtual machine such as vmware or virtualbox, both are free to use and are handy for developers.

edwardkort, 2015-06-12:
WIDesigner uses Jidesoft's application framework to manage look and feel across the supported operating systems and UI managers. And simply using Java's cross-platform L&F does not work reliably for all supported OS/UI-manager combinations using this application framework. And this development team is not prepared to write the entire application interface in plain-vanilla Swing

Jidesoft's application framework claims to support Linux/Gnome and Linux/KDE. Can you try either a Gnome or KDE compliant UI manager?

rabfulton, 2015-06-12:
I tried in linux with xfce in a vm, same result. I also forced java to run with gtk look and feel, same result. Not sure where the problem could lie...

Will have another think about it tomorrow with a fresh head.

edwardkort, 2015-06-12:
Forcing the java L&F will have little or no effect: this is managed internally as WIDesigner is initialized.

If you can, please try Gnome.

edwardkort, 2015-12-27:
Closed due to lack of response.

closed 2015-12-27 by edwardkort