edwardkort/WWIDesigner #34: interesting problem

stanleyloftness, 2015-06-12:
Pat Haran gave me the diagram for one of his G flutes at the Zion flute school.  I made one and it has a excellent extended scale, with all notes within 5 cents.  When I try to recreate it with the WWI designer, I can't get it even close to tuned.  Any idea why?

[image]

edwardkort, 2015-06-12:
Stan,
Please send me the instrument.xml and tuning.xml you created. Pay particular attention to the top-of-the-bore details, hole sizes, how many direction holes, and choke shape.

edwardkort, 2015-12-27:
Closing issue because of no response.

closed 2015-12-27 by edwardkort