edwardkort/WWIDesigner #37: Fixing the position of one or more holes

unclebagley, 2015-09-19:
I am greatly enjoying "playing" with WIDesigner. I have recreated several of my better flutes quite accurately using this marvelous toolkit. I have a question regarding fixing hole locations. Is it possible to preset the location of one or more holes as a constraint set? Thank you.

edwardkort, 2015-09-20:
Barry,

There is no optimizer strategy that allows you to specify the location of an arbitrary set of holes, leaving the other hole positions to be set by the optimizer. I can't even think of how to do this and control for hole-position conflicts. That said, there are several ways to accomplish this with the existing optimizer strategies, depending on your needs.

1. The top hole can be set in a fixed position by choosing the appropriate "Bore top to Hole 6 (top), bore-length fraction". Set the same number in the upper and lower bounds fields.

The next two methods assume that you can get close with the existing optimizer strategies, but just want to move some of the holes to specified positions. So first, do the run with that strategy...

2. Let's supposed that you want to move just one hole and let the optimizer reposition the other holes (this works for any number of holes). In the optimized instrument file, change the position of the hole to your desired value. Then run the "Hole size only" optimizer. Note the hole size for your fixed-position hole. Then choose the "Hole size & position" optimizer, and set the lower and upper bounds for just that hole's size in the constraints, and optimize using either of the last two instrument files. The fixed-position hole should not move much.

3. This is the one that I use frequently. Suppose you run the "Grouped-hole position & size" optimizer and get a tuning that is not quite good enough. But you realize that if you move some of the holes a bit, and angle the holes, you might get a better tuning and still have the holes look equally spaced on the outside of the flute (tuning is based on where the hole is on the inside of the flute). So, in the optimized instrument file, set the hole positions to wherever you please. Then run the "Hole size only" optimizer. The holes will stay where you put them, and the program will change the hole sizes for the best overall tuning.

Hope this helps.

unclebagley, 2015-09-20:
Thank you Edward. I think either method will work for what I want to do. I will give them both a try over the next few days and report back.

Barry

From: Edward Kort 
Sent: Saturday, September 19, 2015 11:06 PM
To: edwardkort/WWIDesigner 
Cc: Barry Gordon 
Subject: Re: [WWIDesigner] Fixing the position of one or more holes (#37)

Barry,

There is no optimizer strategy that allows you to specify the location of an arbitrary set of holes, leaving the other hole positions to be set by the optimizer. I can't even think of how to do this and control for hole-position conflicts. That said, there are several ways to accomplish this with the existing optimizer strategies, depending on your needs.

  1.. The top hole can be set in a fixed position by choosing the appropriate "Bore top to Hole 6 (top), bore-length fraction". Set the same number in the upper and lower bounds fields. 
The next two methods assume that you can get close with the existing optimizer strategies, but just want to move some of the holes to specified positions. So first, do the run with that strategy...

  1.. Let's supposed that you want to move just one hole and let the optimizer reposition the other holes (this works for any number of holes). In the optimized instrument file, change the position of the hole to your desired value. Then run the "Hole size only" optimizer. Note the hole size for your fixed-position hole. Then choose the "Hole size & position" optimizer, and set the lower and upper bounds for just that hole's size in the constraints, and optimize using either of the last two instrument files. The fixed-position hole should not move much.

  2.. This is the one that I use frequently. Suppose you run the "Grouped-hole position & size" optimizer and get a tuning that is not quite good enough. But you realize that if you move some of the holes a bit, and angle the holes, you might get a better tuning and still have the holes look equally spaced on the outside of the flute (tuning is based on where the hole is on the inside of the flute). So, in the optimized instrument file, set the hole positions to wherever you please. Then run the "Hole size only" optimizer. The holes will stay where you put them, and the program will change the hole sizes for the best overall tuning.

Hope this helps.

closed 2015-12-27 by edwardkort