edwardkort/WWIDesigner #40: Windows 10

JRSflyer, 2016-02-10:
Thank you for the response concerning Apple installation.  I have a Windows partition so I installed it in Windows 10 and all the boxes are completely filled with whatever is in them and the description window does not let you see all the wording in the window.  Also all the lettering is very small.

Any suggestions?

edwardkort, 2016-02-10:
I run WIDesigner on a Windows 10 box with no problems. But I have no experience with Mac partitions.

Ideas:
1. Does your partition manager have any UI scaling settings?
2. Windows 10 has several scaling settings: in the System control panel, under Display, you can "Change the size of text, apps, and other items". My setting (my eyes are old) is 125%. Under themes, there are also a bunch of settings for size.

There are no customizable settings in WIDesigner to tweak to change component sizes. Hope the above helps.

Cheers,
Edward Kort

bhp1, 2016-02-10:
When I run WWIDesigner on a 1280x800 display screen, I find I need to shrink the height of the console box to leave more room for the edit panel before the instrument and tuning dialogs will display well.

A display screen at least 960 or 1080 pixels high is preferred.

edwardkort, 2016-02-27:
No additional feedback, so I am closing this.

closed 2016-02-27 by edwardkort