edwardkort/WWIDesigner #41: Lost menu bar

JRSflyer, 2016-02-11:
Last issue.  I am staying with the Apple version since i am not sure how well the partition will work.  

In listening to the You Tube video #3, I clicked to close the menu bar when you showed us how but now i cannot get it back.  Any suggestions?

edwardkort, 2016-02-11:
If you only closed the menu bar (but still have the tool or standard bar), just right click (option click on the Mac, I think), and reselect the menu bar in the list of checkboxes that appear.

If you have closed all the bars, it is more interesting. On Windows, WIDesigner creates a file called "default.layout". It puts this file where other apps write their data. On my computer, it is in the folder:
C:\Users\[user name]\AppData\Roaming\Woodwind Instrument Designer
Delete the file and WIDesigner will come up in its default layout.

On a Mac, with any luck, the file has the same name. Do a search, including hidden and protected files.

Without a menu bar or tool bar, there is no place we could have put a reset command.

Cheers,
Edward Kort

JRSflyer, 2016-02-11:
no luck does one of the other developers know the file name?

edwardkort, 2016-02-11:
None of the developers work on a Mac.

Do a search for a folder name starting with "Woodwind". It should be in a parent folder where other applications store their preferences. On the PC, the folder is named "Woodwind Instrument Designer"; there may be underscores instead of spaces on other operating systems. Then delete the contents of that folder.

In the next release, we will put a command-line argument that allows resetting the layout.

Cheers,
Edward Kort

edwardkort, 2016-05-19:
Fixed. See commit message above. Will be part of the next release, 1.2.0.

edwardkort, 2017-01-11:
The original fix threw a null-pointer exception only on the Mac.

edwardkort, 2017-01-11:
Fixed in v2.0.1 with the modification in behavior (no configuration pane) noted above.

edwardkort, 2017-01-11:
Closed with release v2.0.1.

closed 2017-01-11 by edwardkort

JRSflyer, 2017-01-16:
In windows, the words and numbers overlap so only maybe 60% of the word or
number is visible.  I tried changing window font sizes and screen sizes
with no change.

Any suggestions?

rick

edwardkort, 2017-01-16:
Rick,

Can you attach a screen shot showing this behavior? And what is your display size %?

JRSflyer, 2017-01-17:
Here is the photo.  It looks the same at 100% and 200%  The computer is A Surface Pro.

edwardkort, 2017-01-17:
Rick,

Your screenshot did not come across. In order for it to show, you have to use the GitHub Issues page and drop the image file in the Write box; email doesn't do this. And since I closed the "Lost menu" issue, this will require starting a new issue. Alternatively, just email me directly: edward.kort@gmail.com

Cheers,
Edward