edwardkort/WWIDesigner #43: Add defensive code so study models aren't used on inappropriate instrument types

bhp1, 2016-05-19:
Flute and fipple-flute calculators are not suitable for reed instruments.  Now that reed instruments are supported, some defensive code is advisable so calculators and optimizers aren't used on instrument types that they won't work on.

edwardkort, 2016-12-29:
Let's look at structural modifications that enforce such cohesion, such as specific interfaces supporting each instrument type. Compile-time errors are much better than run-time exceptions.

bhp1, 2017-01-02:
When a user has selected one study model, they can still open instruments of the wrong type.  Not clear how we could control this without a run-time exception, either at open time or at calculation time.  Personally, I'd prefer the exception at calculation time.

edwardkort, 2018-04-24:
No unit tests?

bhp1, 2018-04-26:
New unit tests are somewhat limited, because src/test doesn't yet _have_ any non-fipple instrument examples: the flute and reed calculations are still only beta versions and we do not have enough real data to test against the functionality.  For the purposes of the unit test, I fabricated a reed instrument.

edwardkort, 2018-11-21:
Included in v2.2.0 release.

closed 2019-07-29 by bhp1