edwardkort/WWIDesigner #45: Add taper optimizer for instrument bells

bhp1, 2016-05-19:
Brass instruments and some reed instruments terminate in curved horns or bells.  A bore model built from short cones can model a bell accurately enough for tuning purposes.  However, we'll need a new optimizer (or two) to optimize the profile of these bells, and the taper of any conical segment leading into the bell.  Such an optimizer could also be used for creating a bell model from scratch.

bhp1, 2016-08-14:
BoreDiameterObjectiveFunction and BorePositionObjectiveFunction checked in.  Some changes are still called for so they can play nicely with others in a merged objective function.

bhp1, 2016-11-22:
BoreObjectiveFunction and HoleAndBoreObjectiveFunction, with their corresponding Global versions to use the DIRECT optimizer, complete the set of objective functions for optimizing instrument bore profiles.

edwardkort, 2017-01-11:
Closed with release v2.0.0.

closed 2017-01-11 by edwardkort