edwardkort/WWIDesigner #48: Fingering Pattern

JRSflyer, 2016-06-24:
Ed  Finally getting the hang of the software.  
Suggestion in the next revision:  When we create a fingering pattern in the tuning wizard it would be nice to have the weight column so we can enter our weighs and save them with the pattern.  Every time we open our tuning files, we have to add the same weights since we use the same ones 99% of the time.  It would be a time saver.
Thanks for your time.

Rick

edwardkort, 2016-06-24:
Rick,

When you open your tuning file in the Editor pane, and then make changes (weights, fingering, frequency, or name), just go to the File menu and do a Save or Save as. Those changes will now be persisted and come up the next time you open the tuning file.

We did not expose weights in the Tuning Wizard (instead setting a default value of 1) for several reasons:
1. The Wizard interface was already very cluttered
2. Most users would always want a weight of 1 for each fingering
3. Only the NAF study exposes weights. Changing the weights for a Whistle study would have unintended consequences
4. Changing weights seemed more of an on-the-fly thing while optimizing a flute. I manipulate the weights a lot when optimizing a particular design, but those changes are unique for that flute - I don't save them
5. The Save/Save as capability in the Editor pane gives you full persistence capabilities.

Cheers,
Edward Kort

JRSflyer, 2016-06-24:
I was having no problem with the software, made all my tuning files, and
was able to get a fipple factor when I was learning the software.  I am
 tried to do a fipple factor again and followed the directions and the save
window will not come on.  I have clicked new, added note, and 440, and
named the file with a description but the save but the save button remains
unlit.  any suggestions?

edwardkort, 2016-06-24:
Rick,
After you enter the 440 frequency, you have to either hit ENTER or click on another cell on that fingering line. Otherwise, the program doesn't know you have finished editing - to activate the Save button. You can tell that the 440 has been accepted: the program will append .0000 to your value.

Cheers,
Edward Kort

closed 2016-08-17 by bhp1