edwardkort/WWIDesigner #49: hemi question

JRSflyer, 2016-06-25:
Thanks for the help the last few days.  One more question.  Do you have different measurements for the distance the hemi bore end is from the north end of the TSH or do you use 0.125" for all of the bore diameters.

edwardkort, 2016-06-25:
Headspace (bore above the TSH) is a matter of personal preference. I want as small a headspace as possible, without having to hand cut much wood at the top of the TSH. I find that 0.125" works for most bore diameters. For small flutes, I might increase this a little.

The longer the headspace, the harder it is to get the second octave notes in tune - at least for my designs. I know other makers who use a headspace length of half the bore diameter, so that the TSH lies entirely on a "flat" section of the bore.

Cheers,
Edward Kort

JRSflyer, 2016-07-06:
Ed
Got the 440 and 432 tuning programed but was wondering how to program for
the diatonic scale?
rick

edwardkort, 2016-07-06:
Rick,
I use the same workflow, in the Tuning Wizard, regardless of the scale mode. Let's take an A4 major (diatonic, in the incorrect parlance used with the NAF) scale as an example:

1. In the Scale Symbols page, load a standard "Scientific Symbols, sharps only". This gives an 8-octave chromatic symbol set. Click on the A4 cell (the lowest note we're interested in). Hold down the shift key and click on the C#6 cell (the highest note you might be interested in for the NAF). Click the "Delete unselected" button. You now have a chromatic symbol set for A4 with a limited range (for a NAF). Save it if you like.
2. In the Musical Temperament page, load a standard "12-Tone Equal Temperament". Don't bother reducing its size.
3. In the Scale with Intervals page, click the "Load symbols from Symbols Page" and "Load intervals from Temperament Page" buttons. In the "Symbol List" table, click on the A4 cell; Hit Cntrl (Cmd) - A (selecting the entire list), and drag the list to the "Symbol" column of the "Scale with Intervals" table. Do the same with the "Ratio List" data, putting it in the "Interval" column. Then, in that column, select all the interval cells below the C#6 and click "Delete selected cells". You now have a chromatic, equal temperament interval scale starting at A4.
4. In the Scale with Frequencies page, click the "Load intervals from Scale Intervals Page" button. Set your reference note as desired and click "Create scale with frequencies". You now have a chromatic, A4, equal temperament scale with the desired reference frequency. Give it a name and description and save it; you can use it anytime you want any flavor of A4 scale.
5. Still on the Scale with Frequencies page, edit the "Notes" table. Either select the notes you want and click the "Delete unselected notes" button, or select the notes you don't want and click the "Delete selected notes" button. This is where you would select the notes you want to design for in your diatonic flute. Save the scale, or just load it in the Tuning page.

A diatonic NAF is an interesting design problem. A warning: you will not be able to achieve a well-tuned chromatic NAF with the standard diatonic fingerings (see flutopedia for fingering patterns), so carefully select the notes that are important to you. It helps to use a smaller bore diameter for these flutes, both for the second octave and chromatic tunings.

Cheers,
Edward Kort

closed 2016-12-27 by edwardkort