edwardkort/WWIDesigner #56: Support closing the end of the bore for certain notes

bhp1, 2016-12-27:
Playing technique for some instruments includes closing the end of the bore for certain notes.  For uilleann pipe chanters, the end of the bore is closed on the player's knee for most notes.  For overtone flutes, the end of the bore is closed with the player's finger to get alternate harmonics.

- Add openEnd boolean data member to Termination class, defaulting to true.
- Change termination calculators to return StateVector.closedEnd() if openEnd is false.
- Add openEnd boolean data member to Fingering class, defaulting to true.
- Change Instrument.setOpenHoles to transfer openEnd from fingering to termination.
- Add allowsClosedEnd boolean data member to FingeringPattern class, defaulting to false.
- Change fingering pattern and tuning views to include checkbox for allowsClosedEnd, and
when allowsClosedEnd is true, a radio button for openEnd in each fingering.
- Reed models require that at least one of the holes or the end must be open.

edwardkort, 2016-12-29:
Alternatively, give such instruments their own studies and do not touch any of the existing code. There will be massive code reuse, and the derived classes will be trivial.

bhp1, 2016-12-29:
From our perspective, the difference is not in the instrument but in the tuning.  The tuning includes notes for which the end of the bore is closed.  The external instrument description is no different.  In principle, this is possible with any woodwind: someone could get it into their head to stop the end of a NAF on their knee to get accidentals.

edwardkort, 2016-12-30:
For the NAF, no useable tone is produced when the end is closed until virtually all the finger holes are opened. So I would not want to expose this capability in the NAF study.

bhp1, 2017-01-14:
No need for allowClosableEnd member, or other changes to FingeringPattern.  Add alternate renderer/editor for fingerings with closable end: FingeringWithEndRenderer, FingeringWithEndEditor.  WhistleTuningPanel can use these alternate classes for the fingering pattern column if any of the notes have openEnd data, much as it displays min/max freq or optimization weights only if the Tuning contains them.

bhp1, 2017-01-14:
Resolved in `modelling` branch.

bhp1, 2017-02-07:
Merged with `master`.

bhp1, 2017-09-13:
Released with 2.1.0.

closed 2017-09-13 by bhp1