edwardkort/WWIDesigner #57: Add support for valves and crooks (optional bore sections)

bhp1, 2016-12-27:
Trumpets, etc. use valve-controlled crooks rather than toneholes to alter the acoustic length of the bore.

edwardkort, 2016-12-29:
Is this scope creep? The name of the program is Woodwind Instrument Designer. Do we want to support brass instruments? How about strings?

bhp1, 2016-12-29:
Point taken.  Brass instruments themselves are not high on my priority list.  However, the feature needed to support them--optional bore sections--would allow us to model tuning slides in woodwinds, and it would also simplify the calibration of mouthpieces that have been measured against several sections of plumbing pipe.

edwardkort, 2016-12-30:
So I can then assume this is a very low priority?

bhp1, 2016-12-31:
I wouldn't say _very_ low.  If I find myself with a lot of reeds I want to calibrate, I'll be taking a close look at optional bore sections.