edwardkort/WWIDesigner #59: Save as ... using same name as an existing file locks up application

edwardkort, 2016-12-30:
The Editor panel does not distinguish between two files, with different paths, but with the same file name - in conjunction with a "Save as". In fact, the application no longer functions (confusing operations on the two files) and has to be closed by the operating system (Task Manager in Windows).

The scenario:
1. Open foo.xml from directory A
2. Open bar.xml from directory B
3. Do a Save as with bar.xml, naming it foo.xml, still in directory B

The application is now hosed. The Window menu does not distinguish between the two files; trying to edit either file causes corruption; the application cannot be closed except by externally killing the process.

Just opening two files with the same name causes no problem. 

This is a bug in the underlying JideSoft libraries. A bug has been reported to the company, with no response yet.

bhp1, 2016-12-31:
When you open two files with the same name, the JIDE library assigns a different name to the second, apparently by appending "3", to avoid trouble.  This shows up in the name used in the Study panel, or as the default name when you use File-->Save as.

edwardkort, 2016-12-31:
So it appears that the internal renaming is invoked on an Open but not on a Save as. Should be an easy bug to fix, but still nothing back from JideSoft.

edwardkort, 2017-02-01:
On 1/31/2017 JideSoft released a new version of the JDAF library that resolves this issue. However, they neither updated the version number nor the release notes. So just download the latest release. I will incorporate the new library in the next version, now to be v2.1.0, which will also include the latest JIDE libraries.

bhp1, 2017-02-06:
Not impressed by the fix.  Two issues are immediately obvious:
When you open a file, the entry added to the study tree pane includes the full path name of the file, not just the simple file name.  When you do "Save As..." on a file with a name like myTuning.xml, the save dialog comes up with the file name set to myTuning.xml.xml.

edwardkort, 2017-02-06:
For the first issue, full path showing in the study view, that is unfortunately our problem. JideSoft set the dataModel name to the unique full path name of the represented file, which was a necessary change in the presence of files with the same simple name (and probably the only change they made). It is now our responsibility to separate out the display name and the unique name in the study tree. I will take that on.

How do we want to distinguish two files with the same simple name in the tree? As opposed to the editor tabs, where you can see the associated file content, selecting from the tree does not give that information. Alternatively, we can just leave the display as is, showing the full path.

I haven't looked at the second issue. That is probably a Jide bug - a side effect of using the full path name for the dataModel name. I'll look at it in the next few days.

edwardkort, 2017-02-06:
After looking at the code, I propose that the study view always show the simple file name, but hovering the cursor over this name shows a tooltip with the full path name. And a believe I have a pretty simple way of doing this. Opinions?

bhp1, 2017-02-06:
I'm in favour of your proposal.

bhp1, 2017-09-13:
Released with 2.1.0.

closed 2017-09-13 by bhp1