edwardkort/WWIDesigner #60: Can't open 2.0 on MAC

stanleyloftness, 2017-01-05:
Version 1.1 works fine.  I installed the latest JAVA.  I can open 1.1 but not 2.0

bhp1, 2017-01-05:
Neither Edward nor I have a Mac, so we won't be able to re-create this issue.  Any sign of an error message indicating what's going wrong?

stanleyloftness, 2017-01-05:
It says it can't launch the program and to check the console for error messages.  Checking the console, I can't find anything specific.

Maybe someone else with a MAC will have an idea.

Stan

edwardkort, 2017-01-05:
Stan,

If you wouldn't mind helping us track this down ...

I am assuming you currently have both WIDesigner-1.1 and WIDesigner-2.0 folders.

1. From the WIDesigner-1.1/lib-1.1 folder, copy the 10 files starting with "jide-".
2. Paste those 10 files in the WIDesigner-2.0/lib-2.0 folder. Allow the OS to overwrite the existing files.
3. Try to run WIDesigner 2.0.0 again.

Let us know if this fixes the problem.

Thanks,
Edward

stanleyloftness, 2017-01-05:
No, unfortunately that didn't change anything.  I get the same message.

Stan

edwardkort, 2017-01-06:
Stan,

Those JIDE libraries are the only ones we changes between versions. And we really haven't done an significant UI work.

Please (if you are willing) reboot your computer - in case any java processes are still running. Try to run v2.0.0. Then send us the exact wording of the error message and whatever the console shows, even if it doesn't seem relevant.

Thank you,
Edward

stanleyloftness, 2017-01-06:
Rebooting the computer makes no difference.  Here is the error message.  The program tries to launch and then quits.  I can't figure out what to look at in the console.

Stan

edwardkort, 2017-01-06:
Stan,

The error message didn't come through. As for the console (assuming we're looking at the java console or a command-line window), there should be many lines of stack trace. We need all of those lines. They are not easy to understand, but we should be able to ferret out what is causing the invocation exception.

Thanks,
Edward

stanleyloftness, 2017-01-06:
I checked the open console tab in JAVA, but it does not show up.

Stan

edwardkort, 2017-01-07:
Stan,

With no informative error message, and no Macs among the development team, we are stymied. Perhaps someone else will have the same issue, but with more information.

If you are using WIDesigner exclusively to design NAFs, v1.1.0 will continue to work well for you. The only NAF-specific changes in v2.0.0 relate to rare optimizations that have many solutions with nearly equal tuning accuracies.

The next time you get an OS or JRE update, please try v2.0.0 again. I still believe that this issue is caused by some strange interaction between MacOS, java, and the UI libraries in WIDesigner. And if I get any updated libraries, I would like to recruit you to see if they resolve this.

Thanks for your help and patience,
Edward

stanleyloftness, 2017-01-07:
Thanks for trying. WI 1.1 works wonderfully.

Stan

philwestwell, 2017-01-09:
Hi Stan, 

I installed the code on my Macbook over the weekend so I'll have a look at this today. 

Phil

philwestwell, 2017-01-09:
Hi again, Stan, I got a reproduction on this (hopefully the same) issue. Looking into it now.

edwardkort, 2017-01-11:
Fixed in v2.0.1.

stanleyloftness, 2017-01-11:
Wonderful, works perfectly. Thanks

Stan

Reopened #60<https://github.com/edwardkort/WWIDesigner/issues/60>.

-

edwardkort, 2017-01-11:
Closed with release v2.0.1.

closed 2017-01-11 by edwardkort