edwardkort/WWIDesigner #65: Cannot add holes to a tuning file without starting from scratch

edwardkort, 2017-03-31:
If you want to add holes to an existing tuning file, you must use the Tuning Wizard to build a tuning file.

edwardkort, 2017-03-31:
For flutes and whistles, the ability to add holes is now included in the tuning file editor of the main application. Because this is a very rare scenario for NAFs, this ability has not been exposed for the NAF study.

However, in the Tuning Wizard, additional holes may be added to the foot of an instrument for an existing tuning file (or a de novo one). For the NAF, this scenario is more common: adding tuning/direction holes to an existing design.

Will be included in the next release: 2.1.0

bhp1, 2017-09-13:
Released with 2.1.0.

closed 2017-09-13 by bhp1