edwardkort/WWIDesigner #66: Missing Edit -> Options

milancosnefroy, 2017-08-05:
Hello,

"Options..." seems to be missing from "Edit" menu here.

I have tried locating the problem, but couldn't find any clues.

java -jar WIDesigner-xx.jar returns no errors.

OS: Ubuntu 17.04 with MATE Desktop
JVMs tried: openjdk 8 and 9, oracle java 8
WWDesigner versions: 2.1, 2.0.1, ..... 1.0

Would you have any idea?

Thanks,
Milan

milancosnefroy, 2017-08-05:
Workaround: as I expected, works properly with Win32 version of Oracle Java in WINE.

edwardkort, 2017-08-05:
Hello Milan,

The menu layout in WIDesigner attempts to comply with the different operating-system UI standards. For Windows, preferences are found in the "Option" menu item under "Edit". For other operating systems, preferences may be found in the "Preferences" menu item under "File". I don't know what the MATE paradigm is under Linux. But just look around: it's there.

Cheers,
Edward

closed 2017-08-30 by edwardkort