edwardkort/WWIDesigner #67: Fipple factor

bijlevel, 2017-09-11:
Probably because I'm rather new to the subject of NAF, but I don't seem to grasp the meaning of the fipple factor. I did watch tutorial 9 a few times, but still it is not clear to me. It means (quoting Edward Kort): " how much shading of the TSH is accomplished by the bird geometry". So when I have a TSH length of 4.57 mm and a fipple factor of 0.75, does this mean that 25% of the TSH length is covered by the bird or (most unlikely) 75%? Sorry for asking something which probably is obvious...

edwardkort, 2017-09-11:
The "fipple factor" is, I believe, unique to WIDesigner. There is no formal acoustic theory that addresses the complexity of the fipple geometry and its effect on tuning. The fipple factor is my attempt to do so in a way that:
- is simple
- is amenable to empirical determination (through optimization)
- relatively stable across similar fipple geometries
- accurately predicts tuning in real-world tests
- makes some (at least in my mind) logical sense.

Internal to the program, the effective TSH diameter (the circle with the same area as the rectangular TSH) is multiplied by the fipple factor. This diameter is used in all further calculations.

So a geometry with a measured fipple factor of 0.75 is equivalent to one having a TSH with a 25% smaller equivalent diameter. If the TSH width is constant, this is saying that the TSH is essentially 25% shorter in length (north to south).

Hope this answers your question.

Cheers,
Edward Kort
Wood Wind

bijlevel, 2017-09-11:
Thank you for your quick reply! I'm not sure I understand it. At the end you say:" ... the TSH is essentially 25% shorter in length ...". So I think shorter than what? Does it in fact mean that when cutting out the actual TSH on the flute I should make it 25% longer OR should I use the calculated dimensions from WID, because all the other outcomes (holes diameter, spacing and position etc.) are based on this TSH geometry and fipple factor?

bhp1, 2017-09-11:
Another way to look at it: the fipple factor is an attempt to capture all effects of the mouthpiece geometry that we _cannot_ measure, to calibrate the WIDesigner mouthpiece model with a real mouthpiece.  It might be better to treat the scale of values as arbitrary, not proportional to anything in particular.  

Before you build a mouthpiece, use an estimated fipple factor or one that works for a similar mouthpiece.  After you've built a mouthpiece (before drilling any toneholes), measure actual playing frequencies, and use the fipple factor calibrator to adjust the fipple factor so that the model agrees with the measurements.

bijlevel, 2017-10-31:
Finally I think I know how this works, thanks to your input! I did build a flute, still without holes, and it turned out that with a flute length, measured from the TSH top, of 317 mm the root note was Bb4. This seemed inconsistent with the calculations, because with this length the root note should be A4. But this was the case with the calculated values (by WID) for **TSH width** and **length**, and the **flue depth**. I changed these values to the values measured from my actual flute. I did repeatedly change the **Fipple Factor** and ran each time the optimisation until WID came up with the exact length of my flute. At the end the Fipple Factor was 1,2. BTW I did use a tuning for a Bb4 flute with a pentatonic scale). So I hope I did it right, because now I will drill the holes according to the outcome of WID. (I did use "grouped hole position and size".)

Cheers,
Sjoerd

edwardkort, 2017-11-01:
Sjoerd,

There is an optimizer strategy called (surprise) "Fipple factor" that does all of the fipple factor changes automatically. The video tutorial that demonstrates this is at:
https://www.youtube.com/watch?v=FdsGqeKcp4U&feature=youtu.be

Cheers,
Edward Kort
Wood Wind

bijlevel, 2017-11-02:
Edward, thanks for the reply. I did watch the tutorial now with more understanding of WID. I know I need to watch it more often ;-), but I think at the end I will understand it all. But as you said, the learning curve is quite steep.

All the best,
Sjoerd

closed 2018-04-23 by edwardkort