edwardkort/WWIDesigner #71: Allow copy, cut, paste in Name and Description fields of tuning wizard panels

bhp1, 2018-04-14:
At present, the tuning wizard panels use NoOpTransferHandler for all Name and Description fields.  The intention is to prevent dragging cells from tables elsewhere on these panels and dropping them on the text fields.  Unfortunately, this disables all copy/cut/paste functionality in these fields.

There is no public text transfer handler that we can subclass to disable copy/drop from tables while retaining other copy/cut/paste functionality.  The default javax.swing.plaf.basic.BasicTextUI.TextTransferHandler is private.

I suggest we remove the use of NoOpTransferHandler, retaining the default non-public TextTransferHandler.  This will restore copy/cut/paste, including (unfortunately) copy/drop from tables.  Copy and drop from tables will work; although I don't see it doing anything particularly useful, it won't do any particular harm.  By way of comparison, Name and Description fields in InstrumentPanel use the default TextTransferHandler.

edwardkort, 2018-04-15:
I just ran the tests by commenting out the setTransferHandler calls in configureDragAndDrop for the Name and Description widgets in a couple of wizard component panels. Drag and drop still shows the cursor disallowing drops in these widgets if you are dragging a non-text data or multi-column data. However, if you drag from a text field (like the Scale Symbols table), the cursor shows drag/drop is allowed, but nothing happens (this is my "inconsistent behavior" problem).  But copy/paste works just fine.

So if you know a way to get rid of the incorrect drag/drop cursor behavior, go for it.

bhp1, 2018-04-15:
Have you pulled the latest updates?  The latest updates use proprietary data flavors to give more consistent transfer behaviour both within the application and from WIDesigner to external applications like Excel.  When I commented out the setTransferHandler calls, drag and drop to Name and Description fields produced expected behaviour in all cases, even when dragging multi-column data (not necessarily useful, but predictable).  Without the latest updates, I would expect a ClassCastException when you tried to drag or copy from a text column to a text field, though it may have been caught in TextTransferHandler.

(While drag and drop doesn't work _between_ Name or Description fields in the wizard, I now see it being useful to copy and paste from Name to Name or Description to Description.)

edwardkort, 2018-04-16:
I seem to be getting different behavior than you, or we have different expectations of "predictable behavior".

I pulled the latest code. Then I commented out the setTransferHandler calls for nameWidget and descriptionWidget in ScaleSymbolListPanel.

Go to the Scale Symbols page in the Tuning File Wizard and load a standard symbol set. Select one or more rows in the Symbol List table. This data can now be dragged on dropped onto either the Name or Description fields. To me, this does not seem like desirable behavior. Preventing such behavior was the intent of the NoOpTransferHandler class.

bhp1, 2018-04-16:
Yes, that's the behaviour I would expect.  It is exactly what you see if you select cells from an Excel spreadsheet and drag them into the Name or Description field of an InstrumentPanel.  We'd get the same behaviour with the default transfer handlers of InstrumentPanel if we turned on drag from its tables.  (Try copy-and-paste from an InstrumentPanel table to the Name or Description.)  It may not be useful, but it follows UI conventions.

edwardkort, 2018-04-19:
Drag a fingering onto the Name field (after commenting out its NoOpTransferHandler). You get a representation consisting of X and O characters. There is nothing in the interface (XML or otherwise) that has such a fingering representation.

The purpose of DataFlavor is to prevent dropping an apple onto a field expecting oranges. Your proposal prevents the Name and Description fields from even evaluating the presented DataFlavor. 

There must be a solution out there that maintains the integrity of text fields. We didn't write the first drag/drop interface that contained text fields.

edwardkort, 2018-04-20:
Created a NoDropTextTransferHandler to replace the NoOpTransferHandler.

edwardkort, 2018-11-21:
Included in v2.2.0 release.

closed 2019-07-29 by bhp1