edwardkort/WWIDesigner #72: A little help please

Tlbtgraham, 2018-05-02:
This probably isn't the way to handle MY problem, but here I go with apologies to everyone.

I'm trying to make a C4 Diatonic flute with a thumb hole and for the life of me, I cant seem to get the program to work for me.  Here are the details;

TSH Width = .438
TSH Length = .191
Flue Depth = .03
Diameter = 1.15
Flange = 1.44
Fipple Factor = 1.019
Flute length is 22"

The fundamental is a strong 261 at 22".  I have tried to put in all the tuning and fingering for a 7 hole flute, but I just cant seem to get things right for the hole placement and optimized tuning.  

Any help is appreciated.  

I also want to express my appreciation for this program and the tutorials.  Amazing job.  Thanks for the help.  

Tom

edwardkort, 2018-05-02:
Tom,

Not a lot of information here. The likely culprit is an "impossible" set of fingerings. Zip up the instrument, tuning, and constraints files that you are using and attach them to your reply. Also ensure that you are using the latest release of WIDesigner.

Cheers,
Edward Kort
Wood Wind

Tlbtgraham, 2018-05-02:
Wow, thanks for the very quick response.  

I'm not sure I am zipping the files you mentioned correctly so I took a screen shot of each and zipped those too.  I hope this helps, if not, I'll need a little more help on zipping the actual files I think.  

Thanks again for the great help.  

Tom

edwardkort, 2018-05-02:
Tom,

The zip files did not come across. You need to drag them from your file system to the Comment box you are writing to.

The files I am requesting are XML (text) files. I'll walk you through bundling them (although I don't know what operating system you are using):
1. Create a directory/folder on your desktop. Call it anything you want.
2. In WIDesigner, open your instrument, and make sure it shows in the editor pane (the big box on the right).
3. Under the File menu, do a Save As. Save it to the folder you made in step 1.
4. Do steps 2 and 3 for your tuning file.
5. Do steps 2 and 3 for your constraints file. If you hadn't saved it, select the optimizer you are using, recreate it, and then do step 3.
6. Zip up the folder you made in step 1.
7. Compose a new comment and drag the zip file to that comment.

Screen shots of the interface don't do me much good. I need the XML.

Cheers,
Edward

Tlbtgraham, 2018-05-03:
Perfect.  I’m going to be tied up for a couple hours with my Granddaughter.  Umm jump on it when we return.  
Thanks. 

Tom

Tlbtgraham, 2018-05-03:
OK, lets see if this worked.  

Tom

edwardkort, 2018-05-03:
Tom,

No joy. Try attaching the zip to an email:
edward.kort@gmail.com

Cheers,
Edward

Tlbtgraham, 2018-05-03:
How about this?

Tom

Tlbtgraham, 2018-05-03:
Or maybe this?

Tom

edwardkort, 2018-05-04:
Tom,

There is still no file in your comments. Please just attach it to an email, at the address I gave above. There might be a permission issue for you on GitHub.

Cheers,
Edward

Tlbtgraham, 2018-05-06:
[Kort.zip](https://github.com/edwardkort/WWIDesigner/files/1977260/Kort.zip)
[Files for Edward Kort.zip](https://github.com/edwardkort/WWIDesigner/files/1977261/Files.for.Edward.Kort.zip)

Tlbtgraham, 2018-05-06:
Edward, did this last effort work?  

Tom

edwardkort, 2018-05-06:
Tom,

Yes, you got it this time. I will look at the files and optimization and get back to you. It might take me a few days.

Cheers,
Edward

Tlbtgraham, 2018-05-15:
> Using the files from kort.zip above, with no modifications, a respectable solution was obtained with WID version 2.1.0 (see the attached zip file). No further action is indicated pending a response from the user.
> Results_ENK.zip

Thanks Edward

Tom

Tlbtgraham, 2018-05-15:
Thanks for the help.

closed 2018-05-15 by Tlbtgraham