edwardkort/WWIDesigner #75: Fipple Factor on tapered bores

cyberhek, 2018-08-21:
At what point do I determine the fipple factor with tapered bore NAF flutes? I couldn't find an explanation in your tutorials.  The starting and ending points of the tapered bore, I would guess, are determined on an instrument starter flute first and then after construction, the FF is determined and then the final hole placement is calculated using the same starting and ending taper dimensions.  Is this correct or am I missing something?  I really enjoy tuning my flutes with your program.

edwardkort, 2018-08-21:
What you describe is an appropriate way to determine the FF. I use a somewhat more involved approach:
1. I keep an historical list of all FFs for each style of flute I make. Within a style, even for different flute keys, the FFs are quite stable.
2. I use the appropriate historical FF to design the flute.
3. I then make the flute according to the design.
4. Before glue the two halves together, I measure the bore profile. No matter how careful I am, there is always some differences between design and as-built.
5. I then enter the as-built bore profile in WIDesigner.
6. I glue up to halves, craft the bird and splitting edge, and voice the flute.
7. Then I cut the blank until the fundamental is very close to being in tune.
8. At this point, I calculate the as-built FF with a design file containing no holes, but a bore length and profile equal to the flute in hand.
9. With this FF, I then calculate the holes.

A lot of steps, but no surprises.

cyberhek, 2018-08-22:
Great description. Thanks so much for the info.

closed 2018-08-22 by cyberhek