edwardkort/WWIDesigner #76: double declaration in java fixes

poulphunter, 2018-09-02:
I have compared FEM results for brass instruments with this algorithm, I've seen some differences.
After some investigation, the double declaration was'nt always well declared in Java.
With these modification, my results where better :)

poulphunter, 2018-09-02:
can you add me rights to make a pull requests (create branch / pull requests) to push my modifications ?

edwardkort, 2018-09-03:
For non-collaborators, the pull request process is best for all. Using git commands (or your favorite git client):

- git clone \<our URL>
- cd \<project>
- git checkout -b \<name of your topic branch>
- (make your changes)
- git commit
- (go to WIDesigner GitHub page and Fork) or git remote add \<your fork name> \<our URL>
- git push \<your fork name> \<your topic branch name>
- (on the WIDesigner page, click Pull Request) or git request-pull origin/master \<your fork name>

Then, the collaborators can look at the changes and do the appropriate merge.

poulphunter, 2018-09-04:
Thanks for the detailed process.
I've created merge request https://github.com/edwardkort/WWIDesigner/pull/77

bhp1, 2018-09-04:
Thank you for the copious corrections.  It will take me a while to check over all the details, but I will work on pulling them into the repository.

bhp1, 2018-09-10:
I've applied the explicit double type designator to constants for which it might be significant (those with a fractional part).  Please pull the changes and see if they resolve the discrepancies you found.

edwardkort, 2019-07-29:
No instances of unhandled (by the compiler) upcastings (float -> double, etc.) were found. Nonetheless, explicit double designators were made throughout the code with no changes observed in the calculated values.

With no further information received from the user, the issue is now closed.

closed 2019-07-29 by edwardkort