edwardkort/WWIDesigner #79: Dumb user questions

jewelmind, 2018-10-06:
Hi, I feel like I'm probably in the wrong place, as this seems like a page for developers, but I can't find a page for users. Please just delete this if user questions are out of place here.

I'm finding WIdesigner very, very opaque. I'm used to using Hans Bracker's whistle calculater, which makes perfect sense to me, so I'm not completely illiterate in these things, but at the outset just getting a simple whistle design done using WIdesigner feels like studying for a degree in computer acoustics. Would be great if someone could simplify it a bit and make it user-friendly, even at the cost of say, resolving everything to tenths of a mm rather than hundredths.

I'm trying to be patient and follow the instructions, but at the moment I feel stymied by the very first instructions in the Whistle Study. It feels like a Catch 22 - I have to have an instrument already designed and made to measure the frequencies, but I can't make the instrument until I design it. What do I do? Is there a default frequency values file somewhere that I've missed? Do I measure the frequencies from a completely different instrument as a starting point?

The same thing appears in the Calibrating section, which says to "Measure the current dimensions of the instrument" as a starting point... I have to have an instrument before I can design it... Catch 22...

Honestly, if I have to make lots of rough approximations of an instrument as part of the design process, I may as well ditch the software altogether. I do want to use WIDesigner as I want to make tapered bores, which the Hans Bracker calculator can't do, but if I'm doing effectively just as much prototyping as a non-software trial and error process, there's no practical reason to use WIDesigner at all.

bhp1, 2018-10-06:
First off, you are in the right place.  We do use the issues list to track development issues, but it also works as a forum for user questions.  

Re-reading the Whistle Study page in the wiki, I can see that it needs work.  I'll have a go at it when I have time.  Meanwhile, I'll try to give a gentler introduction here.

In measuring a bunch of different whistles, I found that the dimensions I could measure didn't entirely explain the tuning of all of them, so the WIDesigner whistle model uses two calibration factors to help the model fit better with the tuning of real whistles.  The first factor is the window height.  In principle, this is a dimension you can measure--how far is it from the inside of the tube to the outside at the window.  In practice, the wall around the window may have different heights around the perimeter of the window, making it hard to settle on a single number.  Because of the uncertainty, we allow for adjusting the `window height` parameter to improve the match between predicted and actual tuning.

If the whistle doesn't exist yet, take a guess at the window height based on the expected height of the window and any walls around it.  If you have a whistle with the same sort of mouthpiece as the one you're building, you can use the calibration process on that whistle, to see what `window height` works out to for that whistle.  If you have a partially built whistle, you can use the calibration process as soon as you can get a few notes out of the whistle.

The other calibration factor is a dimensionless number we've called _beta_.  I've seen beta values of 0.3 to 0.4 in the literature, and used values up to 0.5 or 0.6 to calibrate whistles.  If the whistle doesn't exist, start with a value of 0.4 and go from there.  Again, if you have a similar whistle, or a partially built whistle, you can see what the calibration process turns up.

I should also say something about the frequency measuring process.  The _only_ reason you have to measure minimum and maximum frequencies is to see how well the model is predicting the tuning of your whistle, and for the calibration process to improve the predictions.  My prejudice is to measure often and re-calibrate when necessary.  Others may be satisfied to use initial estimates for window height and beta, and _never_ measure the min/max frequencies.  In retrospect, this is a perfectly reasonable use of WIDesigner, and probably a good place to start with your first whistle.

Does this help?

bhp1, 2018-10-08:
I'll see what I can do about your suggestions for improved documentation.  Meanwhile, the next thing you'll need is a tuning file that gives the frequencies you _want_ your whistle to play.  For almost all of the WIDesigner tools, you need to select both an instrument and a tuning.  If you are working on a high D whistle, the WIDesigner release includes sample Whistle Study tuning files for D5-Equal (equal temperament) and D5-Just (just intonation), as well as samples for Bb4, A4 and D4 (low D).  If you're more ambitious, you can create your own tuning with the tuning wizard.  Once you have a tuning file, you can generate the tuning table, and see how close you are.  

As long as your starting design is close enough that WIDesigner is able to find a playing frequency near every note in your tuning file, you can select an optimizer and click on the Optimize button.  I suggest you start with a simpler (and faster) optimizer like hole size and spacing before you try introducing a taper.  Each of the whistle optimizers has built-in default constraints, so you don't need to create explicit constraints unless you want something other than the defaults.  After running an optimizer, generate another tuning table to see how close you got.  You can also look at the tuning graph for a visual depiction of the tuning, but it may take some practice to understand what you're seeing on the tuning graph.

Let me know how that goes.

jewelmind, 2018-10-08:
Okay, many thanks again. I'll give all that a go and get back to you.

bhp1, 2018-10-09:
Updated the Using WIDesigner wiki page with more specific guidance for new users.

edwardkort, 2018-11-21:
Noted in Release Notes for v2.2.0.

closed 2020-12-09 by edwardkort