edwardkort/WWIDesigner #81: In NAF Study, calculated hole sizes are smaller than the tuned holes

edwardkort, 2019-02-03:


edwardkort, 2019-02-03:
The hole model in the NAF Study has a hole-size multiplier, meant to approximate a resistance factor. Currently, the value of this multiplier is hard wired in the code. It should be exposed to the user, and an optimizer strategy created to determine its value.

edwardkort, 2019-02-03:
Burton, all good points. To add a bit more detail to the observation:
- The flutes in question are chromatically tuned - only one optimum position for each tone hole
- The bore length and hole positions are correct, just the hole size differs from calculated
- Fipple factor has been optimized.

Thus, the mouthpiece model appears not to be at issue.

The current DefaultHoleCalculator, used by all the studies, already has a hole-size calibration factor. Unless explicitly set in its constructor, it defaults to 1. Only the NafStudyModel changes this default, and sets it to a constant of 0.95 based upon personal measurements.

I propose the following:

1. Add an optional element to the Instrument model - a hole-size multiplier
2. Change the signature of the StudyModel.getCalculator() method to pass in an Instrument instance. Every place it is currently called, that instrument instance is available.
3. Only in the NafStudyModel will the getCalculator method create a DefaultHoleCalculator with the hole-size factor extracted from the instrument
4. For instruments read by the NafStudy with a null factor, the factor will be set to the historical value for backwards compatibility. Otherwise, the read value will be used
5. Create an optimizer strategy in the NafStudyModel to determine the multiplier value and insert it into a new instrument instance.

What do you think?

bhp1, 2019-02-05:
Your proposed procedure would freeze the hole size multiplier at calculator creation.  There would be no possibility of optimizing (calibrating) it because any changes to the instrument wouldn't get propagated to the calculator.  To calibrate the factor, it would need to be passed to HoleCalculator.calcTransferMatrix, either in Hole or as a separate parameter.

If I understand it, WIDesigner's prediction of the tuning of the as-built flute has all the notes except the bell note being flat.  Is this correct?  If so, I still wouldn't rule out the possibility of correcting this systematic error with a change to the mouthpiece model.  I ran the latest A4 instrument file you sent me through the NAF model and the whistle model.  For A#5, the NAF model predicted A#5 closed would be almost 20 Hz sharp of A#5 open; the whistle model predicted the two notes would be within 2 Hz of each other.  Which is closer to the truth?  If both fingerings are in tune, this suggests the prediction _without_ the hole size adjustment (whistle model) works better for that note.  

If there is a problem with the tonehole model, perhaps the problem is in the thickness rather than the diameter.  NAFs generally have thicker toneholes than the ones I've been working with; perhaps the model loses accuracy for thicker toneholes.

edwardkort, 2019-02-05:
Burton,

Good catch. We would need to expose the hole-size multiplier to the calculation framework, rather than just make it an attribute of the HoleCalculator constructor.

For the A4 flute, I just pulled it out. The difference in pitch between the two fingerings of A#5 is about 30 cents - closer to the NAF model. However, it is virtually impossible to blueprint one of my finished flutes for the validations we need to run: all of the toneholes have been shaped for voicing and tuning. At a minimum, I asymmetrically round the inside of each hole. So the exact cylindrical diameter and position to center cannot be measured with any certainty.

When Antoine and Scavone derived the tonehole model, wasn't Antoine using it to design saxophones? The tonehole height on a sax is typically quite large. Roughness of the tonehole might be a factor, but I make mine very smooth.

I think we need several reliable blueprints to test against (instrument and tuning files). It's still too cold for me to go out to my shop to make flutes. I will put up a request on the forum that initiated this discussion.

edwardkort, 2019-06-12:
I built optimizers for the finger-adjustment factor and hole-size multiplier. Using about 8 test flutes, I performed a large number of optimization runs. For a single flute, optimizing for these two additional parameters reduces the tuning error. However, there is a larger variation of these optimized parameters, between flutes, than I am comfortable with.

For now, I vote not to expose either parameter to the user with an optimizer strategy. Nonetheless, the average optimized hole-size multiplier value is slightly larger than currently used in the NAF study. I have changed the default value in the DefaultHoleCalculator, for the NAF study only, in the HEAD revision.

bhp1, 2019-07-22:
WIDesigner had a tendency to predict sharp for notes with mostly closed toneholes and flat for notes with mostly open toneholes.  DefaultHoleCalculator now has an adjustment factor for the finger intrusion into the hole, to allow for the reduction in closed tonehole volume from the presence of a finger.  Default value set to give an adjustment comparable to what Paul Dickens used in his 2007 thesis.  This adjustment reduces the tendency to predict sharp for closed toneholes, although there are still residual prediction anomalies.  Adding further adjustments in DefaultHoleCalculator, or making the finger adjustment stronger (smaller value) doesn't do much to eliminate the remaining prediction anomalies.

edwardkort, 2019-07-22:
For the aggregate of the NAF tests flutes I evaluated (8 flutes), using fixed values for the finger-intrusion correction and hole-size multiplier, the best fit was obtained with NO finger-intrusion correction. This conclusion might change with better test flutes. But for now, the NAF calculator will use no finger-intrusion correction.

edwardkort, 2019-07-29:
Although a Resolved label has been added, this issue will remain open.

closed 2019-09-08 by edwardkort