edwardkort/WWIDesigner #82: Alternative fingering

stanleyloftness, 2019-04-26:
First, I love WWID,  I recommend it to anyone building flutes.  I am trying to build flutes with a alternative fingering pattern of Pat Haran.  I can't seem to find the parameters to alter that will get my deviation below 9.  I am looking to suggestions.  

Stan
[TUNING DIAGRAM MULTIMODAL.pdf](https://github.com/edwardkort/WWIDesigner/files/3122029/TUNING.DIAGRAM.MULTIMODAL.pdf)

edwardkort, 2019-04-27:
Stan,
You're not providing much info (key, bore, hole size/spacing constraints, etc.). I just did a run with your provided fingering for an A4 flute, bore 0.94" (no hole grouping): average tuning error 3.2 cents.

stanleyloftness, 2019-04-27:
Using your 0.875 starter and no hole constraints I get this
[A4 NAF Multimodal.pdf](https://github.com/edwardkort/WWIDesigner/files/3123941/A4.NAF.Multimodal.pdf)

edwardkort, 2019-04-27:
That's a pretty "amusing" result. The only issue is see from the screen shot is the placement of the TSH: part of it is actually above the bore. But I don't think that would cause the tuning error you are seeing. Since the devil is in the details, please zip up the instrument, tuning, and constraints xml files and attach them to your next comment. Then I can try to confirm and investigate your results. Thanks.

stanleyloftness, 2019-04-27:
How do I create a zip file with these?

edwardkort, 2019-04-27:
If you don't have a zip utility, just drag the separate files to the bottom of the comment box. Hopefully GitHub won't try to interpret the XML (some interfaces do).

stanleyloftness, 2019-04-27:
[Archive.zip](https://github.com/edwardkort/WWIDesigner/files/3124155/Archive.zip)

stanleyloftness, 2019-04-27:
Did that work?

edwardkort, 2019-04-27:
Thanks for the files, Stan.
If I just check the tuning using your provided files, the average deviation is 18.96 cents, not the 190 cents in your screenshot. I used 22.222 degrees C, 45% humidity. Running the "hole size and position" optimizer only brought it down to 18.30 cents deviation.

Then I looked at your tuning file and it is pretty strange: no C5 (which is normally just Hole 1 open), no G5, an A5 with the fingering expected for a G#5, etc. If this is your intended tuning, without errors, then realize that there are many such tunings that impose extreme challenges.

stanleyloftness, 2019-04-28:
This is a tuning from Pat Haran.  He calls it an alternative tuning, some of his clients call it multimodal.  It has an extended range, and the scale ascends by uncovering one hole at a time.  There is less cross-fingering.  It might be easier to learn on this tuning.  Amazing Grace plays perfectly.  I am trying too make a A, G, and F flute with this tuning to experiment.  I usually make choked flutes, with a choke length of about 0.25 inches. (Easy to rout). I can get the deviation down to about 9, but the second octave notes are way out.  Do you hav ed any suggestions on how to bring the second octave into tune.

Thanks Stan

edwardkort, 2019-04-28:
I don't believe that this fingering pattern can yield very accurate tuning across the 1-1/2 octave chromatic scale. Even if I introduce a bore taper, the best average tuning deviation I get is about 12 cents.

You have a couple of options:

1. Increase the weights of those notes that are the most out of tune. The average deviation will go up, but you won't have wildly divergent mistunings
2. In a stepwise fashion, set the weights to 0 for those notes that are least important. You may find that a small number of notes are driving the mistunings.

The "normal" chromatic fingering pattern on a NAF plays Amazing Grace with no difficult fingers, too. What you are missing is the major second, which must be half holed. I will often make 7-hole flutes, with the bottom two holes almost in line (and covered with one finger) - like a recorder: no half holing needed and quite easy to play.

stanleyloftness, 2019-04-28:
I have also tried seven  hole flutes with good success.  I am interested in what things affect the upper octave the most.

edwardkort, 2019-04-28:
Lowering the bottom holes will raise the pitch of the second-octave notes, and increasing the headspace (the bore above the TSH) will lower them. The first aspect is handled by the hole position optimizers (within your allowable constraints). I haven't made an optimizer for headspace, because typically you want that as small as possible. For your application, that may not be true.

And of course varying other parameters (TSH length, bore diameter, etc.) will affect the ability to voice the second-octave notes. The note spectrum graph can help you evaluate that aspect.

stanleyloftness, 2019-04-29:
Thanks,

Stan

closed 2019-06-12 by edwardkort