edwardkort/WWIDesigner #84: Blank Document

bluesbi, 2019-05-25:
When i tryed to open an instrument on linux LTS it just show a blank document without info and the console shows this:

Exception in thread "AWT-EventQueue-0" java.lang.NoClassDefFoundError: javax/xml/bind/JAXBContext
	at com.wwidesigner.util.BindFactory.unmarshalXml(BindFactory.java:103)
	at com.wwidesigner.util.BindFactory.isValidXml(BindFactory.java:219)
	at com.wwidesigner.gui.StudyModel.getCategoryName(StudyModel.java:463)
	at com.wwidesigner.gui.util.FileOpenDialogPreviewPane$XmlPreview.configure(FileOpenDialogPreviewPane.java:120)
	at com.wwidesigner.gui.util.FileOpenDialogPreviewPane$1.propertyChange(FileOpenDialogPreviewPane.java:70)
	at java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:341)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:333)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:266)
	at java.desktop/java.awt.Component.firePropertyChange(Component.java:8728)
	at java.desktop/javax.swing.JFileChooser.setSelectedFile(JFileChooser.java:506)
	at java.desktop/javax.swing.JFileChooser.setSelectedFiles(JFileChooser.java:539)
	at java.desktop/com.sun.java.swing.plaf.gtk.GTKFileChooserUI$SelectionListener.valueChanged(GTKFileChooserUI.java:457)
	at java.desktop/javax.swing.JList.fireSelectionValueChanged(JList.java:1804)
	at java.desktop/javax.swing.JList$ListSelectionHandler.valueChanged(JList.java:1818)
	at java.desktop/javax.swing.DefaultListSelectionModel.fireValueChanged(DefaultListSelectionModel.java:219)
	at java.desktop/javax.swing.DefaultListSelectionModel.fireValueChanged(DefaultListSelectionModel.java:186)
	at java.desktop/javax.swing.DefaultListSelectionModel.setValueIsAdjusting(DefaultListSelectionModel.java:723)
	at java.desktop/javax.swing.JList.setValueIsAdjusting(JList.java:2152)
	at java.desktop/javax.swing.plaf.basic.BasicListUI$Handler.mouseReleased(BasicListUI.java:2958)
	at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:298)
	at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6632)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
	at java.desktop/java.awt.Component.processEvent(Component.java:6397)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5008)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117)
	at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190)
	at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.desktop/java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at java.desktop/java.awt.Dialog.show(Dialog.java:1070)
	at java.desktop/javax.swing.JFileChooser.showDialog(JFileChooser.java:756)
	at java.desktop/javax.swing.JFileChooser.showOpenDialog(JFileChooser.java:653)
	at com.jidesoft.app.framework.gui.FileDialogRequestHandler.openFileDialogSwing(Unknown Source)
	at com.jidesoft.app.framework.gui.FileDialogRequestHandler.showDialog(Unknown Source)
	at com.jidesoft.app.framework.gui.ApplicationDialogsUI.showDialog(Unknown Source)
	at com.jidesoft.app.framework.gui.filebased.FileHandlingFeature.openAs(Unknown Source)
	at com.jidesoft.app.framework.gui.filebased.FileHandlingFeature.openDataFiles(Unknown Source)
	at com.jidesoft.app.framework.gui.filebased.OpenFileAction.introduceData(Unknown Source)
	at com.jidesoft.app.framework.gui.actions.DataAction.actionPerformedDetached(Unknown Source)
	at com.jidesoft.app.framework.gui.actions.OpenAction.actionPerformedDetached(Unknown Source)
	at com.jidesoft.app.framework.gui.GUIApplicationAction.actionPerformedDetatched(Unknown Source)
	at com.jidesoft.app.framework.gui.GUIApplicationAction$0$1.run(Unknown Source)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
[... 27 lines cut ...]
	at com.jidesoft.app.framework.gui.GUIApplication.a(Unknown Source)
	at com.jidesoft.app.framework.gui.GUIApplication.addDataModel(Unknown Source)
	at com.jidesoft.app.framework.DesktopApplication.openData(Unknown Source)
	at com.jidesoft.app.framework.gui.GUIApplication.openData(Unknown Source)
	at com.jidesoft.app.framework.gui.filebased.FileHandlingFeature.openDataFiles(Unknown Source)
	at com.jidesoft.app.framework.gui.filebased.OpenFileAction.introduceData(Unknown Source)
	at com.jidesoft.app.framework.gui.actions.DataAction.actionPerformedDetached(Unknown Source)
	at com.jidesoft.app.framework.gui.actions.OpenAction.actionPerformedDetached(Unknown Source)
	at com.jidesoft.app.framework.gui.GUIApplicationAction.actionPerformedDetatched(Unknown Source)
	at com.jidesoft.app.framework.gui.GUIApplicationAction$0$1.run(Unknown Source)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

Can anyone help me ?

edwardkort, 2019-05-25:
The error indicates that whatever file was read was neither an instrument, tuning, or constraints xml file: it is failing trying to read the first element of the xml.

So, first a few questions:
1. Are you trying to open an unmodified sample instrument included with the distribution?
2. If not, can you open one of the samples? Please try an instrument and a tuning.
3. Have you successfully used WIDesigner under Linix (any version of either)?

Thanks,
Edward Kort

bluesbi, 2019-05-25:
Just an add, i opened a tunning file and an instrument but it seems like i can't open any kind of optimization and constraint, cause the option it is unavaible. Here it goes an print.

 
[image]

Sorry my english.

bluesbi, 2019-05-26:
1 No
2 Yes i was trying to open a sample who came with the software.
3 I tried some versions but still the same error.

edwardkort, 2019-05-26:
Your screen shot doesn't look like I would expect. Our UI libraries explicitly support Gnome and KDE. I don't know whether LTS is a flavor of either. Do you happen to have either Gnome or KDE installed and can try them?

bluesbi, 2019-05-26:
I installed an Kubuntu KDE as you asked, but still the same problem.

edwardkort, 2019-05-26:
Thanks for trying KDE. One last simple UI test:
1. Follow the instructions in UNIX_README.txt (in the distribution) for invoking WIDesigner, but add the following argument:
   -usekde=true
2. Try to open a sample file
3. Attach a screenshot of the interface to your comment.

bluesbi, 2019-05-26:
When i tried happened the same, but i had this print on the shell:

[image]

bluesbi, 2019-05-27:
Maybe i could put this code by another way ? or try by another way ?

edwardkort, 2019-05-27:
The additional information from the last run shows that the third-party libraries used in implementing WIDesigner's UI do not support Linux under the flavors you have tried. We validated WIDesigner under CentOS, but have not done any further Linux testing. There might be further issues beyond the UI ones, but without a functional UI, there is no benefit in trying to pursue them.

So until the library provider (JideSoft) updates the libraries, WIDesigner will not run under your Linux version. Sorry. And JideSoft has been very slow in updating their UI libraries. I still have outstanding bugs reported to them 2 years ago.

bluesbi, 2019-05-27:
Allright, what version of linux i can install  ?

bluesbi, 2019-05-27:
Im realy want to use this software and even in the future help with the software tests.

bhp1, 2019-05-27:
Based on our limited experience, CentOS, Red Hat, and possibly Fedora would be your best bets.

bluesbi, 2019-05-27:
THX if not work i will open another schedule.

edwardkort, 2019-05-28:
Have you tried WINE or another Windows emulator under Linux?

I will contact JideSoft and ask about "reference" Linux implementations. What hardware are you running?

bluesbi, 2019-05-29:
Im not tried the Wine because im now installed the CentOS to try it.
Im running on an desktop.

bluesbi, 2019-05-29:
I used the software on the CentOS right now and its running just fine, but i didn't like the system at all he just no work for me. I will try the Fedora now.

bluesbi, 2019-05-29:
Maybe i should try just a theme on the CentOS, i will think about it for now, but on this system its working, im just not used with him.

bluesbi, 2019-05-30:
Yes of Fedora doesn't work either. I will try by another way

bluesbi, 2019-06-12:
In the end i just find a way out using another OS, i will keep using windows for use just the software

closed 2020-12-09 by edwardkort