edwardkort/WWIDesigner #89: can't use WIDesigner with Mac OS 10.15

stanleyloftness, 2019-10-16:
I recently installed Mac OS 10.15, Catalina, and now WIdesigner won't open

edwardkort, 2019-10-17:
Hi Stan. I don't own a Mac. But I've been reading a lot of reports of software incompatibilities under Catalina. And most of the recommendations are, "Don't upgrade the OS."

You don't provide much information. Try bringing up a command window (whatever it's called under Mac OS) and invoke WIDesigner with a command line (see the UNIX_README for details). That way you will see OS-level exceptions.

I suspect that Java has problems with Catalina. The above should tell.

And there is no Mac OS-specific code in WIDesigner, so there is likely nothing that can be done on the programming level.

Please let us know what you find.

Cheers,
Edward Kort

stanleyloftness, 2019-10-18:
I reverted back to the previous operating system, works fine.

Thanks Stan

closed 2020-12-09 by edwardkort