edwardkort/WWIDesigner #90: L/D ratio of 20

stanleyloftness, 2020-01-21:
How is it possible to make a Native American flute with a L/D ratio of 20 or above.  It would seem that the length of the flute would be determined by the fundamental, bore diameter, and TSH size.

edwardkort, 2020-01-21:
Hi, Stan. To a first approximation, you are correct. However, changing the bore diameter does change the length of the flute for a given pitch and TSH geometry. For example, k1 (the virtual extension at the foot) changes, because it is about 1/3 of the bore diameter.

Give it a try in WIDesigner. Create a flute with no holes and one fingering. Run the "Hole size and position" optimizer and note the bore length. Then change the bore diameter and run the optimizer again. You will see that the bore length changes - as does the L/D ratio.

I still use NAFlutomat to estimate L/D ratios. I haven't incorporated this metric explicitly in WIDesigner.

stanleyloftness, 2020-01-22:
I tried it with a A4 tuning and a .625" bore diameter and it came up with a 13.438" bore length.  I'll make it and see how it sounds.  Thanks Stan

edwardkort, 2020-01-22:
Stan, let us know how it works out. I have used large bores in my A4 flutes for many years: 0.875". But my next flutes will have somewhat smaller diameters. Pretty much every time I change my flute design, I will make a series of flutes with different bore diameters to assess playability and tone quality.
Cheers, Edward

stanleyloftness, 2020-05-05:
I made a series of flutes with small bore diameters
A4 .625" bore 13.25" length
F#4 .75" bore 15.8" length
F4 .75" bore 17.2" length
All have straight bores.

They all in tune and sound decent, a very different quality to the sound, hollower.
The finger holes are very small and require a bore thickness of .25 to make them reasonable.
The overflow more easily and require more controlled breath pressure. Nodal interference is a problem if the 6th hole is closer than 27%
The second octave is supported somewhat better.
I couldn't design one with a tapered bore to see how that would sound

Stan

edwardkort, 2020-05-05:
Hi Stan,

 

Thanks for the update. I’m surprised that they sound hollow. My smaller-bore flutes tend to sound buzzier. Perhaps your TSH is too long (N/S) relative to this new geometry. A little bit of plasticine (clay) on the north end is an easy way to determine this. And I like the touchier nature of a small-bore flute – more opportunity to mess with the notes. But most of my customers don’t have the breath control necessary to play them well.

 

I have a couple of A4 flutes started on the bench: a 7-hole major scale and a 7-hole (double bottom hole) standard scale. The terminal bore diameters are 0.74”, smaller than my typical 0.875”. The major-scale flute is straight bore, the other has a slight taper. It will be interesting to see how I like the tone and playability.

 

Cheers,

Edward Kort

logo-2_mid-tiny

stanleyloftness, 2020-05-06:
Maybe hollow is the wrong word, they are definitely buzzier.

I am trying to make a rim blown anasazi flute out of some left over 1" ID PVC pipe.  It has a good clear sound of D# and overblows twice.  It is 22" long.  I can;'t seem to find a way of estimating starting hole placement for a six hole anasazi tuned flute.  Do you have any ideas?

Stan

edwardkort, 2020-05-06:
Assuming a fixed embouchure, which is not strictly the case, and end-blown flute is just a fipple flute (NASF or whistle). Every time I try to verify this, I'm hindered by my poor playing technique. I would think that the Whistle Model would give you better results than the NAF Model: it has a window optimizer (similar to the fipple-factor optimizer) that accounts for the multiple octaves, as well as a beta optimizer that accounts for different blowing pressures when playing those notes. The documentation for the Whistle Model may be found in the wiki pages (not the videos).

If I could play better, I would have incorporated an end-blown model long ago. I do like a flute with a variable embouchure for its expressivity.

bhp1, 2020-05-06:
I'd suggest trying the flute model rather than the whistle model.  The calibration factors for the flute model are the airstream length and beta factor.  For an end-blown flute, the airstream length might make a better calibration factor than the whistle model's window height.  You might have to try different values for the [blowing level](https://github.com/edwardkort/WWIDesigner/wiki/WIDesigner-Options#blowing-level) to see what gives the best predictions after calibration.   Start with a blowing level of 9.

stanleyloftness, 2020-05-06:
I can't find the flute calibrator and optimizer in the study panel.  I must n to be looking in the right place.

Stan

bhp1, 2020-05-06:
Go to the Options dialog (Edit->Options) and select the Flute Study on the General Study Options tab.
[image]

Select the blowing level on the Whistle/Flute Study Options tab.

stanleyloftness, 2020-05-09:
Thank you very much for your help, on the Mac OS the options dialog is in the preferences menu.

I would love to help support all the wonderful work you are doing.  I would love to purchase one of your flutes or donate something to a cause you cherish.

Thanks Stan

[image]<https://user-images.githubusercontent.com/1645842/81230591-c54e7400-8fbf-11ea-8893-22886077e70c.png>

edwardkort, 2020-05-10:
Stan, you are welcome. I'm pleased that you find the program valuable.

The programming was a joint effort between me and Burton Patkau. For my part, that people are using the program is reward enough. Burton is welcome to chime in.

Cheers, Edward

bhp1, 2020-05-12:
I'm with Edward, happy to see people using the program and reporting on their experiences.

closed 2020-12-09 by edwardkort