edwardkort/WWIDesigner #91: DockableFrame warnings appear on console

bhp1, 2020-03-05:
When you click or double-click on the header bar of the Console frame or the study view frame, warnings like those below appear on the console.

Mar 04, 2020 8:45:59 PM com.jidesoft.docking.DockableFrame$1 focusGained
WARNING: DockableFrame(Console):focusGained
Mar 04, 2020 8:45:59 PM com.jidesoft.docking.DockableFrame$1 focusLost
WARNING: DockableFrame(Console):focusLost

edwardkort, 2020-03-05:
Supposedly this was fixed in JIDE 3.7.8, released 12/2/19. The current build uses JIDE 3.7.7. The JideSoft changelog reads: "Remove some unnecessary logs of focusLost/Gained in DockableFrame". The functionality of the current WIDesigner release is not impaired.

edwardkort, 2020-05-10:
Fixed in v2.5.0.

closed 2025-01-01 by bhp1