edwardkort/WWIDesigner #92: Severe end choked flutes

stanleyloftness, 2020-07-27:
Edward,

I decided to try make some flutes with severe end shoes like a quena, shakuhachi,or anasazi flute.  I had no trouble designing them on the program, and made a G F# and F flute.  The F and F# flute for example goes from .875 to .5 inch for the last 1/2 inch.  After I built them, when cut to length they all ended up a half step below what was predicted.  Any idea what is going on?

Stan

edwardkort, 2020-07-27:
L
Edward

stanleyloftness, 2020-07-28:
I ran a fipple factor one my F# flute with no holes, which is actually a F flute.  It ended up being .599.  There is something about that severe an end taper that altering the acoustical prosperities.  All three flutes ended up actually being 100 cents flat.
[image]

edwardkort, 2020-07-28:
If I understand your post, your normal fipple factor (0.750) went to 0.599 with the choked flute. Since the fipple factor is a catch-all parameter which includes geometry as well as blowing characteristics, this is not surprising. You should use this new "historical" fipple factor when designing other such choked flutes. But, since cutting the flute to yield an F# flute will cut off all of your choke, I suggest just making an F flute with it.

The real question is whether the tuning offset you are seeing is due to a reduced blowing pressure for the fundamental note. If so, that reduction is likely to disappear as you go up the scale (because the choke will no longer be in play). The NAF model may not accommodate these dramatic and discontinuous changes in blowing pressure, so start with well-undersized holes. After you have made your first flute, you might blueprint it (max and min playing frequencies for each note) according to the protocol for the Whistle model - which will better handle these pressure changes.

stanleyloftness, 2020-08-08:
It turns out that when cutting down the flute to tune the fundamental, 1/4" changes the tuning one half step.  I built a F and F# flute, they sound ok, you do have to alter blowing pressure somewhat, nothing dramatically different in sound.  Interesting experiment.

Stan

closed 2020-12-09 by edwardkort