edwardkort/WWIDesigner #93: Need pointers on how to achieve more robust optimization

Sonoton, 2020-09-24:
Thank you for making a useful acoustic tool, based on published acoustic research, and making it with a great interface.
My issues, perhaps due to my own operator error, are as follows:

1)  I do not have a Constraints Tab in the Study Window when I create a blank or default constraint,  (the program just lists a new row under the Optimizer Tab)
2)  The optimization for me appears to be limited.  I can only get to about  -5 to +5 Hz or 2 to 15 cents note errors on a D4=293Hz flute/whistle (Cmajor scale with Even Temperament), even by weighting the notes with larger deviations, and then only an improvement of half a percent occurs, even by invoking different optimizers successively, after 20 tries.

3) It would be nice to be able to vary the height of the holes.  
4) It would be nice to vary a Whistle's bore profile with steps, rather than entering them manually or using a taper.
5) It would be nice to have the optimizers opt for larger diameter holes during optimizaton to achieve higher, (or even selectable) cutoff ratios.
6) It would be nice to have some user control over the coarseness of optimization, perhaps this could be done by allowing only optimizing a certain section of the instrument along its length.

Any assistance would be appreciated.

bhp1, 2020-09-25:
1.  This one concerns me most, so I'll deal with it first.  If you close all the instrument and tuning files you have open, open one instrument file and one compatible tuning file, select a simple optimizer like hole size optimization, and select File-->Create default constraints, then you should see a constraints form appear on top of your current instrument or tuning, with a new tab on top called Untitled1.  I gather this isn't happening.

Are you using the latest version, 2.5.0?  (I'm not sure it matters, but it helps us when we try to reproduce the problem.)  What platform are you running on?  (We develop and test on Windows.  We've tried it on Linux and Mac, but don't have regular access to those platforms.)

bhp1, 2020-09-25:
2.  Are you trying to optimize over more than one octave, perhaps with some cross-fingerings like OXXOOO for C-natural?  If so, you can't expect to get zero deviation over the whole range.  There are inevitable compromises between C# and cross-fingered C-nat, and across the octave break, C#-D-E, and between low, middle, and high D.  Getting within 15 cents is very good.  (In reality, I can't promise that the accuracy of the model is any better than 5 or 10 cents.)

3.  The right-most column of the instrument's hole table is the hole height, from the inside of the bore to the outside of the hole.  For holes bored through a simple tube, this is the thickness of the tube at the hole.  You can enter a different value for each hole.

4.  For stepped bores, I generally enter two bore points very close together, with different inside diameters, for example, an inside diameter of 17.5 mm at position 60.0 mm, and another with an inside diameter of 18.0 mm at position 60.1 mm.

bhp1, 2020-09-25:
5.  This goes back to your first point.  If you can get Create default constraints working, you will have a form on which you can change the upper limit on hole size for each individual hole.  For example, you may want to allow hole B2 to be up to 11 mm, and leave the rest below 9 mm.  You can even make the limit larger than the inside diameter of the bore, which is physically impossible (and not a useful thing to do).

6.  I'm not entirely clear on your objective here.  Perhaps you could give an example of how you want to use this.  If you want to optimize only some dimensions and leave others unchanged (like because you've already drilled three of the holes), you can create new constraints (back to your first point again) and enter very narrow bounds for the measurements you don't want to change, like hole size between 6.49 and 6.51 mm.  I wouldn't recommend making the upper and lower bounds equal, because some of the optimization functions don't deal well with that.

bhp1, 2020-09-25:
6, again.  You can use bore point names to tell the Upper Bore and Lower Bore optimizers what part of the bore to leave unchanged.  For Upper Bore optimizers, the highest bore point left unchanged is the lowest bore point with a name that contains the word "Head".  For Lower Bore optimizers, the lowest bore point left unchanged is the highest bore point with a name that contains the word "Body", or the lowest bore point with a name that contains the word "Head".

Sonoton, 2020-09-26:
Yes, what you stated in your reply occurs. The new constraints form is created along side the instrument file and tuning file that were opened, and the Untitled1 constraint is listed in the lefthand (Flute) Study window as just another optimizer: there is not a Constaints Tab under the (Flute) Study window, in contrast to what is shown in video 2 Woodwind Designer Installation at time 16:40 to 17:20. Additionally, if I try to instantiate such a Constraints Tab, by going to the Constraints-Flute Study and opening the AirstreamLengthDefaultConstraints.xml for example the program just opens this constraint in my Flute Study window as an .xml file under the Optimizer Tab. In other words, the Constraints tab seems to have disappeared from the program operation in both WWID 2.2.0 and 2.5.0.

It may be that a startup file under Roaming directory in Windows 7 has been corrupted and I should remove all the WWID installations and then reinstall.

Sonoton, 2020-09-26:
2> I was trying to optimize an 8 hole flute, with one of the holes being an octave-type hole. No cross fingerings at all, but the octave reach may have been problematic in terms of the number of variables or the dependencies on the the open holes, which might be equivalent in difficulty.

3> Point taken. This will work.

4> Point taken. I can section the bore as you indicate.

Thank you for these suggestions.

 
 

   
   -

Sonoton, 2020-09-26:
5> I think this is a useful fix. I believe what I was looking for might be under the Multistart Optimization. I will try to educate myself about this.

6> Your suggestion is the fix I was looking for.'

Thank you for your help.

 
 

   
   -

Sonoton, 2020-09-26:
6again> Thank you for explaining this. From your description, may I infer that I could have Headpt1 Headpt2 Headpt3 e.g. for Headjoint Optimizer and Bodypt1 Bodypt2 Bodypt3 e.g. for Lower Bore Diameter Optimizer? 

(I might should have been reading the source code to try to figure out how your objective functions are used in the optimizers, but after extensive reading and watching the videos, I decided to just jump in and start swimming.)

Sonoton, 2020-09-26:
1> I found this by looking at YouTube comments under the WWID videos. It would be a good idea for me to read the entire Wiki, since one viewer indicated that the Wiki provides the solution explicitly: "Unlike the NAF Study Model, other study models do not have a separate branch in the Study tree for Constraints."

So we can consider this issue as closed. 

Thank you for your patience.

bhp1, 2020-09-26:
1. Yes, I get now what you weren't seeing.  In the flute and whistle models, you're not just creating new constraints, you are creating a new instance of the optimizer. When you optimize with those constraints, you're using the original optimizer with the new limits.

bhp1, 2020-09-26:
You can label a single bore point as Head, or Head-Body.  The Upper Bore optimizers will optimize everything _above_ that bore point, the Lower Bore optimizers will optimize everything _below_ that bore point.

closed 2020-12-09 by edwardkort