edwardkort/WWIDesigner #97: Cannot get WWIDesigner to run on windows 10.

Ryanfisher8, 2020-10-01:
hello everyone,

I currently have the java 8 SE download, and have tried running the designer many times, the jar file doesn't work and neither does the command shortcut. I'm probably doing something wrong but have no idea, can someone briefly explain the download process, perhaps i have the wrong version of java downloaded or something else very simple i have overlooked. Running windows 10. any help is greatly appreciated as I am dying to make some flutes. Thanks in advance. - Ryan

edwardkort, 2020-10-01:
Hi Ryan,

Let's take this one step at a time. The first step is to confirm that your java JRE installed correctly. The download URL is:
[https://www.oracle.com/java/technologies/javase-jre8-downloads.html](url)
and you want to download the 64-bit version:
jre-8u261-windows-x64.exe

Run the installer above that you downloaded (you might need to reboot after the installation). Then you need to validate the install. Bring up a command prompt window (in the Start search box, just type cmd and hit ENTER). At the command prompt, type:
**java --version** and hit ENTER.
You should see information about your java installation if everything is correct. Otherwise, you will see:
_The term 'java' is not recognized ..._

If the installation is correct, you should be able to invoke WIDesigner by double clicking the jar file or running it explicitly under java on the command line (see the INSTALL_README.txt).

If still no joy, we'll go on to step 2.

Cheers,
Edward Kort

Ryanfisher8, 2020-10-02:
Hey Ed,

I was running the wrong version of java it seemed, i had tried SE8 and SE7, so after uninstalling those, I installed the correct version, the one you stated above, version jre-8u261-windows-x64.exe.
I then followed your instructions to validate the install, unfortunately you were correct, after typing in java--version, i received the message " Java--version is not recognized as an internal or external command, operable program or batch file."
And just to be sure Ed, after this, i did restart my computer and try it again and received the same message. 
What should we try next Ed?
Thanks for your help,
-Ryan

edwardkort, 2020-10-02:
Hi Ryan,

I may not be the best person to help get your java installation running, mine is so horribly convoluted (java 7-12, jdk and jre), but let's plow forward. You might also do some web surfing to look for recommendations.

The java configuration for command-line invocation and double-click/browser utilization are very different. So let's try the double-click mode. Open a file manager window and navigate to where you downloaded WIDesigner. Double click the file _WIDesigner-2.5.0.jar_. WIDesigner should open in its own window.

Let me know what happens and we'll go to step 3.

Cheers,
Edward

Ryanfisher8, 2020-10-02:
Ed,

previously before installing your recommended java version, clicking the jar file opened a window that was just filled with a bunch of code, but I just clicked it after installing the correct version of java and rebooting and WWIDesigner has succesfully opened for me! Thank you very much, hopefully now I will be able to navigate the designer with the help of your youtube videos, thanks again.

-Ryan

edwardkort, 2020-10-02:
Excellent, Ryan. Now we won't have to get into the arcane world of environment variables.

In addition to the youTube videos, please look at the relevant sections of the wiki on this site.

Cheers, and happy designing,
Edward

closed 2020-12-09 by edwardkort