edwardkort/WWIDesigner #98: Exception when running NAF hole size and position optimizer with no holes

edwardkort, 2020-10-18:
When running the NAF hole size and position optimizer with no holes, the only optimizer in the NAF Study Model that can just change bore length, an exception is thrown. This is because the BOBYQA optimizer is called with just one variable.

bhp1, 2020-10-18:
HoleFromTopObjectiveFunction is a merged objective function, HolePositionFromTopObjectiveFunction plus HoleSizeObjectiveFunction.  You don't want HoleSizeObjectiveFunction involved if there are no holes.  In using HoleSizeObjectiveFunction, HoleFromTopObjectiveFunction assumes there are holes, and so it can safely set the optimizer to BOBYQA.  I suggest you expose HolePositionFromTopObjectiveFunction as a separate optimizer in the GUI.  It doesn't assume there are holes, and uses CMAES if there aren't any.  (Powell might be a better choice, but CMAES should work.

edwardkort, 2020-10-18:
Hi Burton,
HoleFromTopObjectiveFunction is currently only use in the NAF study. It is also one of the few (if only) ObjectiveFunctions that does not check the number of variables before invoking BOBYQA rather than CMAES.

I'll write the test first, and then determine the best way to handle this: new optimizer, smarter HoleFromTopObjectiveFunction, or whatever.

BTW, now that I am making flutes again, I have come across several warts in bore manipulation that I will address in the next couple of weeks.

edwardkort, 2020-10-26:
Fixed. Will be in 2.6.0 release.

closed 2025-01-01 by bhp1