edwardkort/WWIDesigner #102: Import bore points

Prudall, 2020-12-05:
First of all, many thanks for this wonderful program. I was wondering if there is a way of quickly importing bore points, for instance from Excel spreadsheets. I am studing baroque and romantic transverse flutes and the bore points lists are quite long.
Many thanks!

edwardkort, 2020-12-06:
Piece of cake! But there are limitations:
* You need to first create the appropriate number of rows in the instrument bore points table (in WIDesigner)
* You can only copy/paste one column at a time (position or diameter).

To do the operation:
1.  Select the cells in Excel (a contiguous column of individual cells)
2. Copy them (Ctrl-C or Cmd-C)
3. Select the starting cell in the bore points table of WIDesigner
4. Paste them (Ctrl-V or Cmd-V)
5. Do 1-4 above for the second column of values.

Please let me know how this works for you.

Cheers,
Edward Kort

Prudall, 2020-12-07:
Thanks a lot! It works perfectly.

closed 2020-12-07 by Prudall