edwardkort/WWIDesigner #122: Transverse flute

Transverseflute, 2021-02-16:
1) Can you please have some exercise through transverse flute. Fipple  / Beta factor is difficult to find out or may be this program have limitations and not be able to find correct beta factor. I have tried many times but it is not working.

2) I have tried that flute model. I did 2 months practice on it but it behave weird for top three holes. Have anybody built transverse flute using this application? I respect all American flutes but Indian bansuri type transverse flute can play meend which is not possible using any western or American flutes. Indian Classical music is not possible without meend. American flutes and western fluted are jumping from one note to another which is easy to make but flutes having ability to play meend required an extraordinary skill because it also defines the tone in between the notes.

None of flute designing program design the transverse flute correctly.
Please help me out to have right program for transverse flute.
I am using Flutomat online but that program also have some error.
Watch my video to figure out what I want.
My channel:  https://www.youtube.com/channel/UC2MQ08bxGwsqU-gkhpbkTbg

edwardkort, 2021-02-16:
First, your two top questions were moved to the Discussion section under the Flute Study category - and they were answered.

Second, I have not used Flutomat for transverse flutes (for which it was written). When I tried it on Native American flutes, it also gave incorrect answers. That's why I rewrote it, creating NAFlutomat (specifically for NAFs); however, it is not likely to work well for the bansuri, because it will not adequately handle the headspace common in the Indian flute.

Transverseflute, 2021-02-16:
Where is flute study category? Send me a link.

edwardkort, 2021-02-16:
[Discussions/Flute Study](https://github.com/edwardkort/WWIDesigner/discussions/categories/flute-study)