edwardkort/WWIDesigner #126: Just intonation

stanleyloftness, 2021-04-05:
I am trying to make a flute with just intonation.  I created a JI tuning file for A4.  The best I can do in making a flute is a SD of around 9.  Any ideas?

edwardkort, 2021-04-05:
Hey Stan,

I don't know exactly what your flute characteristics, fingering pattern, nor constraints are. Please zip those three files (instrument, tuning, and constraints) up and attach them to your comments.

When I play with a typical geometry, my fingering pattern, and the 16 fingerings I like to get in tune, my best result is about 13 cents average deviation. However, if I leave just 3 notes out of the optimization, I can get the average deviation down to about 3 cents.

Anyway, send me your files.

Cheers,
Edward Kort
Wood Wind

stanleyloftness, 2021-04-07:
[Archive.zip](https://github.com/edwardkort/WWIDesigner/files/6272946/Archive.zip)

edwardkort, 2021-04-07:
Stan,

I concur: with the files provided above, I can only achieve a design with 9.3 cent average deviation. However, if both D#4 and G#4 are left out, a 1.1 cent average deviation is achieved.

Alternatively, you could change the fingerings for D#4 and G#4. Doing so yields a 2.76 cent average deviation (files attached).

Cheers,
Edward
[alt_fingerings.zip](https://github.com/edwardkort/WWIDesigner/files/6274019/alt_fingerings.zip)