edwardkort/WWIDesigner #127: Tuning wizard "Finish" button is misleading

bhp1, 2021-04-05:
From @CamStLouis, the Finish button on the Tuning wizard pages closes the wizard without saving, risking loss of work.

Suggest re-naming it to "Close" or "Quit".  Consider moving it from lower left position.

CamStLouis, 2021-04-05:
And I'd move the "save" button to it's place; it's weird to have your final product saved from up in the middle of the wizard. To me that position kind of implies that it's somehow an intermediate, rather than final, step in making a tuning.

edwardkort, 2021-04-05:
Five of the panes in the Wizard have save buttons. The interface is used to create/modify:
- Scale symbol sets
- Temperaments
- Scales (with symbols and frequencies)
- Fingering patterns
- Tunings (scales and fingerings).

A global save button is inappropriate. And changing the name of the "Finish" button to "Close", IMHO, provides no additional clarity (perhaps less) in the action the button performs.

CamStLouis, 2021-04-05:
I disagree, as the intended final product of the wizard is a complete tuning file. It's ordered in a series of steps, not a series of separate editors, so it follows that the last page would be the most-often-desired final product of the wizard. I don't need to construct or save half of what's in there, I just need the temperament, the fundamental frequency, and the fingering pattern, and then save the final tuning file.

"Finish" on a wizard, in every other program on earth, in the left hand button position we have been trained for as long as computers have existed means "affirmative/save/continue," should not be "close without saving (and no warning, either!)" Freaking muscle memory made me delete tuning files in the past. 

For the text "Finish," I would expect the wizard to close, sure, _but to also open the created file in a tab on the main interface._

bhp1, 2021-04-05:
In my experience with multiple-screen processes, Finish means accepting default behaviour on all the remaining screens.  The screens of the Tuning wizard, for now at least, have no default behaviour, so "accepting default behaviour" doesn't apply.

edwardkort, 2021-04-05:
The "Finish" button name is the default name provided by JideSoft (our library provider), whose business it is to provide OS-compliant UI libraries. I'm not inclined to go against that expertise absent compelling arguments.

CamStLouis, 2021-04-06:
> In my experience with multiple-screen processes, Finish means accepting default behaviour on all the remaining screens. The screens of the Tuning wizard, for now at least, have no default behaviour, so "accepting default behaviour" doesn't apply.

Ahh, I see how that came about. Is there an easy way to make it open the file that's just been created, or is the file only "created" when it's saved?

> I'm not inclined to go against that expertise absent compelling arguments.

Is... is "the end user repeatedly losing all their work after a seven step process" _not_ compelling?

twentyfortysix, 2023-12-07:
I agree, it is misleading, no matter what are the defaults of the library, the UI should be understandable to what it is used for.