edwardkort/WWIDesigner #132: No options/preferences menu item on Linux/GNOME.

olson-sean-k, 2021-07-20:
I'm running WWIDesigner on Arch Linux with GNOME and there is no menu item for options/preferences. As suggested in similar issues, I've looked in places where I'd expect to find such a menu item and have examined all I can find in the UI to no avail.

#32 looks very similar to what I'm seeing now, but it was closed some time ago due to inactivity. In that issue, the JIDE framework was mentioned. Perhaps there's a problem with the framework or how it's being used?

olson-sean-k, 2021-07-23:
> Also, look at discussion item #84. You might want to explicitly invoke WIDesigner with a command-line argument of
-usegnome=true.

Thanks for the suggestion. Unfortunately, the `-usegnome=true` option seems to have no effect.

The Arch Linux machine mentioned above is running GNOME 40. I've tried running WWIDesigner on another machine running Debian with GNOME 3.38 and the menu item appears as `Edit > Preferences`. GNOME 40 brought numerous changes; maybe JIDE isn't compatible?

In the meantime, is there a default layout that can be enabled somehow? I could easily live with a layout that doesn't conform to my desktop environment. :smile:

Thanks again for taking a look.

edwardkort, 2021-07-24:
Sean,

The last changes to the JideSoft UI library we use were made three years ago. So for now, I would say that gnome 3.40 is unsupported.

WIDesigner (through the JideSoft library) stores layout and preferences info in an OS-specific fashion. When you invoke WIDesigner, the first line in the console pane should tell you where that file is found (maybe). It is called default.layout (under Windows). Being binary under Windows, I can't see its contents nor edit it. But it might be text or XML under gnome. Worth a look.

As a longshot, using the gnome 3.38 machine, configure WIDesigner the way you want it. Then copy default.layout to the gnome 3.40 machine. Who knows? And nothing to lose: you can always delete it and have WIDesigner make a new one.

Sorry, but likely no joy.

Cheers,
Edward

Nazrax, 2022-03-04:
Anyone make progress with this? I've also tried both -usegnome=true and -usekde=true to no effect. I'm running Fedora/Mate.

edwardkort, 2022-03-04:
The JideSoft interface library still has not been updated. So ... no progress.

Nazrax, 2022-03-04:
Ok, I've found a workaround:
`export GNOME_DESKTOP_SESSION_ID="this-is-deprecated"` (I don't think the actual value works, but it seems to be traditional).

edwardkort, 2022-03-04:
Nazrax,

Interesting. I wonder whether the jidesoft library is ignoring the command-line variable setting.

If you have a chance, would you try:
export usegnome=true 
before invoking widesigner.

Thanks.

Nazrax, 2022-03-07:
TL;DR: To get a Gnome UI, `export GNOME_DESKTOP_SESSION_ID=whatever`. To get a KDE UI, either `export KDE_FULL_SESSION=true` or add `-Dusekde=true` to the commandline (before `-jar`).

So, `export usegnome=true` doesn't do anything. I _did_ see `usegnome` in the strings, though, so I poked around a little more. Turns out `usegnome` and `usekde` (and `usecp`) are Java system properties it looks for - kind of. 

```
  public static String getVariant() {
    return ApplicationUIManagerFactory.isUseCrossPlatformApplicationUI() ? "CP" : 
    (SystemInfo.isMacOSX() ? "MAC" : 
    (SystemInfo.isWindows() ? "WIN" : 
    (SystemInfo.isLinux() ? (LinuxApplicationUIManager.isGnomeLinux() ? "LINUX_gnome" : (ApplicationUIManagerFactory.isUseKDEApplicationUI() ? "LINUX_kde" : "CP")) : 
    ((SystemInfo.isSolaris() && ApplicationUIManagerFactory.isUseGnomeApplicationUI()) ? "LINUX_gnome" : "CP"))));
  }

  public static boolean isUseCrossPlatformApplicationUI() {
    return Boolean.valueOf(SecurityUtils.getProperty("usecp", "false")).booleanValue();
  }

  public static boolean isUseKDEApplicationUI() {
    return (Boolean.valueOf(SecurityUtils.getProperty("usekde", "false")).booleanValue() || LinuxApplicationUIManager.isKDELinux());
  }

  public static boolean isUseGnomeApplicationUI() {
    return Boolean.valueOf(SecurityUtils.getProperty("usegnome", "false")).booleanValue();
  }

  public static boolean isGnomeLinux() {
    return (Environment.getInstance().getVariable("GNOME_DESKTOP_SESSION_ID") != null);
  }

  public static boolean isKDELinux() {
    return SystemInfo.isLinux() ? "true".equals(Environment.getInstance().getVariable("KDE_FULL_SESSION")) : false;
  }
```

In other words, even though they have a function to check `usegnome`, they forgot to call it! However, `-Dusekde=true` _does_ work. (Incidentally, `usecp` is also honored - though since the crossplatform UI is both the default and broken this isn't particularly helpful).

edwardkort, 2022-03-07:
Yes, unless isUseGnomeApplicationUI (which is public) is called somewhere else. I hate this style of run-on spaghetti code!

Thank you for your detective work. I will update the UNIX configuration recommendations with your work in the next release.

twentyfortysix, 2023-11-21:
This is still problem on both linux and mac, no properties.
while on linux export GNOME_DESKTOP_SESSION_ID=whatever helps, I have no idea what would be the proper equivalent on mac.

There nothing mentioned in the manual, or I have not found it.

Nazrax, 2023-11-21:
The relevant code for Macs seems to be
```
    String str = SecurityUtils.getProperty("os.name", "Windows XP");
...
    if (str.startsWith("Mac OS"))
      if (str.endsWith("X")) {
        _isMacOSX = true;
      } else {
        _isMacClassic = true;
      }  
    }```
So I believe that specifying `-Dos.name="Mac OSX"` as part of the Java command should do the trick.

twentyfortysix, 2023-11-21:
Unfortunately. That does nothing. I try to convince it to use the "native" mac os UI, but without luck.
--systemInfo macOS
export SYSTEM_INFO=macOS
nothing works

it might be that the flag actually is set right, but the app does not react to it and does not modify the UI appropriately and maybe never will... unless they update the library.

[image]

twentyfortysix, 2023-11-21:
btw there is a new version of the library.. might actually fix few things.
http://www.jidesoft.com/history/index.php#3.7.14
If anyone will be able to merge it in to the WID I'll gladly test it :D

Nazrax, 2023-11-21:
Put `-Dos.name="Mac OSX"` _before_ `-jar` - everything after that is counted as a parameter to the app instead of to Java itself.

twentyfortysix, 2023-11-21:
the code you posted triggered error of unrecognized variable
```
Exception in thread "main" java.lang.AssertionError: Platform not recognized
	at sun.nio.fs.DefaultFileSystemProvider.create(DefaultFileSystemProvider.java:70)
	at sun.nio.fs.DefaultFileSystemProvider.<clinit>(DefaultFileSystemProvider.java:37)
	..
```
this actually passed 
`java -Dos.name="Mac OS X" -jar WIDesigner-2.6.0.jar`

although the UI is still incomplete without "properties"

edwardkort, 2023-11-22:
> btw there is a new version of the library.. might actually fix few things. http://www.jidesoft.com/history/index.php#3.7.14 If anyone will be able to merge it in to the WID I'll gladly test it :D

Unfortunately, the JideSoft library that needs updating is the JDAF library, not the various JIDE libraries. And WIDesigner 2.6.0 contains this latest JDAF library. So, no joy from that direction.