edwardkort/WWIDesigner #146: Display errors

merlinusambrosius, 2022-03-08:
I am experiencing errors when navigating the application.  Java is correctly installed.  I am running on Windows 11.  It is a little difficult to explain with words, so I have put together a very short video.  A demonstration of what is happening can be viewed here:

https://rumble.com/vwtutv-wwid-errors.html

As you can see, it is quite unusable.

edwardkort, 2022-03-08:
This is not what you want to hear, but the problem is quite likely to be somewhere in your system. WIDesigner uses no animation nor double buffering.

So, a few questions:

1. Is your video driver up to date and configured correctly?
2. Is your java version current, and what is that version?
3. Do you see this behavior with any other applications, in particular java apps?
4. If you have two screens, do you see this on the other display?
5. Have you cold booted your computer recently? You should.
6. Did WIDesigner ever work on this computer? If yes, what has changed?

That said, I'm still on Windows 10 (Windows 11 doesn't have anything I want and has messed with a number of features I use regularly). But I haven't heard any other user comments to this effect.

Please let me know if you find out anything.

merlinusambrosius, 2022-03-14:
To document the problem, the error was coming up with the Nahimic service in Windows.  To my understanding it is an Asus driver related to Sonic Studio.

edwardkort, 2022-03-17:
Strange stuff!

Nahimic is an audio system that applies EQ curves to known speakers and headphones. If you have an ASUS gamer laptop, there might be a dedicated chip on the motherboard that performs the real-time audio processing. Nonetheless, it "must" have a choosable/configurable playback device driver that you can unchoose/disable on the fly. I have 5 or 6 such drivers on my system.

I haven't used OpenJDK, so I don't know whether it contributes to your display errors. But WIDesigner runs correctly on all versions of Oracle java from 8 through 12.