edwardkort/WWIDesigner #147: Add "Hole and Upper Bore [Global] Optimizer"

Nazrax, 2022-03-16:
I'm trying to create a tapered 3D-printed tin whistle head to fit on a PVC body, and there doesn't seem to be a single optimizer that really suits this task. I've found success by going back and forth between the "Hole and Upper Bore Diameter Optimizer" and the "Hole and Upper Bore Spacing Optimizer," with the occasional "Hole Size+Spacing Global Optimizer" thrown in for good measure. Each run gets me a little closer to the ideal value, and I can skip some steps by guessing at the general direction the optimizers seem to be taking me; but it would be great if there were upper bore optimizers along the lines of the existing "taper" optimizers.

The "Hole and Taper Optimizer" is described as:
> Introduces a basic two-section taper in the bore. Varies the total bore length, spacing between holes, the sizes of holes, the position of the second bore point, and the diameter at the bottom bore point. Keeps the diameter at the first and second bore points constant. Total number of dimensions = 2 * number of holes + 3.

I'd love to see a "Hole and Upper Bore Optimizer" (both local and global) described as:
> Introduces a basic two-section taper in the bore. Varies the total bore length, spacing between holes, the sizes of holes, the position of the second bore point, and the diameter at the _top_ bore point. Keeps the diameter at the _second and bottom_ bore points constant.

bhp1, 2022-03-17:
I think you're going to get better results sticking with the Hole and Upper Bore Diameter optimizer, and adding a few bore points or putting them in different places until you get a design you like, perhaps with sparing use of the Hole and Upper Bore Spacing optimizer. The two-section taper you propose is far more restrictive, and since the bottom section is the cylindrical PVC pipe, there's only so much you can move around that second bore point.

In my experience, going between bore diameter and bore spacing optimizers can be helpful now and then, as you have found, but it isn't always productive, because they can work at cross-purposes.  And trying to optimize bore point diameter and spacing at the same time (beyond the simple two-section taper) can leave you under-constrained, where the optimizer wanders all over the place without a clear global optimum.

bhp1, 2022-03-19:
I can clarify my last point with a simple example.  Suppose we have a 4 cm headjoint with three bore points, one at each end and one in the middle, and suppose the optimum bore is a linear taper from 12 mm diameter down to 10 mm diameter.  If we set the middle bore point at 11 mm diameter, a bore spacing optimizer should put it at 2 cm from the end.  If we set the middle bore point at 2 cm from the end, a bore diameter optimizer should put it at 11 mm diameter.  But where should a bore-diameter-and-spacing optimizer set the middle bore point?  11 mm diameter at 2 cm works, but so does 10.5 mm at 1 cm, or 11.5 mm at 3 cm, or any other point on the linear taper.

Nazrax, 2022-03-21:
If the 2-section taper works for a fixed head with a variable body, then why not the other way around? My first attempt at a tapered head has provided a noticeable improvement over a parallel head, and it tapers from 16 to 12 mm over only 10mm. I would be surprised if there were any benefit to adding another bore point. However, while I would prefer to keep the second bore point closer to the labium (PVC is cheaper than filament), I _can_ put it where ever I need it, since I can add more length to the head and cut length off of the PVC.

bhp1, 2022-03-22:
> I would be surprised if there were any benefit to adding another bore point.

I would be surprised if there wasn't a benefit.  Some time ago, I went through an exercise looking for an optimal headjoint.  Under certain conditions, the optimum looked like a cylindrical insert that narrowed the bore diameter by 1 mm or so, extending about 1 cm or so down from the head of the whistle.

Nazrax, 2022-04-06:
Attaching files as per https://forums.chiffandfipple.com/viewtopic.php?f=1&t=113292&sid=040d5d80a5ac93b85ed1f2818e19ff28&start=15

[taper-files.zip](https://github.com/edwardkort/WWIDesigner/files/8429597/taper-files.zip)

I rotate between the attached Upper Bore Spacing optimizer, the attached Upper Bore Diameter optimizer, the stock "Hole Size+Spacing Optimizer", and occasionally the attached "Gbl Hole Spacing Size" optimizer.

bhp1, 2022-04-06:
That includes the tuning file and constraint files.  Could you supply the instrument file that you applied these to?

Nazrax, 2022-04-06:
Wow, I feel stupid ... I put everything in the zip, but when I reorganized it to get rid of the folders I also got rid of the most important file ..

[High D - Quiet - CPVC - 05 - tapered head.xml.txt](https://github.com/edwardkort/WWIDesigner/files/8429945/High.D.-.Quiet.-.CPVC.-.05.-.tapered.head.xml.txt)
.

bhp1, 2022-04-07:
A few comments about the instrument definition...

As noted in the [Glossary](https://github.com/edwardkort/WWIDesigner/wiki/Glossary), the mouthpiece position for a whistle is the position of the sound blade at the lower end of the window.  With the top of the bore at position 0, the mouthpiece position would be 4 mm, equivalent to the window length.

To use the upper bore optimizers effectively, I suggest you put the label "Head-Body" on the bore point that represents the bottom of the headjoint.  The optimizers will adjust the diameter or position of bore points _above_ that bore point.

You had beta set to 0.2.  Until you have a chance to measure tuning on a real whistle and calibrate the model, I suggest you start with beta at 0.3.

bhp1, 2022-04-07:
A very interesting observation ...

I looked at what would happen with a straight cylindrical headjoint.  With the small window you propose, the optimum hole size and spacing leads to small holes for all but the bottom hole, and the lower octave a few cents _flat_ while the upper octave a few cents _sharp_.  With a wider window, say 5 mm, optimization leads to larger holes in general, and the more conventional situation of the lower octave a few cents sharp and the upper octave a few cents flat.

With the narrow window you propose, I don't think you _need_ any taper in the headjoint to balance the octaves.

[HighD,cylindrical.xml.txt](https://github.com/edwardkort/WWIDesigner/files/8438874/HighD.cylindrical.xml.txt)

closed 2023-01-25 by bhp1