edwardkort/WWIDesigner #154: Varying hole heights can give negative position holes

NetOneGames, 2023-05-25:
[image]

bhp1, 2023-05-26:
I see you are using a custom set of optimizer constraints.  Which optimizer are the constraints based on?  The constraints are what should keep the optimizer from straying out of bounds, so inappropriate constraint settings could easily lead to a situation like this.

NetOneGames, 2023-05-26:
Hi yes custom optimizer , should be in the zip file.   is based on the "hole and lower bore global optimizer " dont think i did anything too wild if i recall just restricted hole sizes to between 7.5mm and 10mm  and the hole spacing to 37mm

bhp1, 2023-05-26:
In the constraints, you've limited the bore length to 600 mm, and allowed the bottom hole, B3, to be up to 600 mm above the lower end of the bore.  When the end of the bore is at 327.69 mm, B3 could be as high as -272.31 mm, which is where it ended up.  That's the thing about the global optimizers: if the constraints numerically allow a configuration, the global optimizer may try that configuration.

At the very least, the sum of the maximum distances between holes in the constraints, including the maximum distance from B3 to the end of the bore, must be less than - significantly less than - the bore length.  The global optimizers work best with narrow constraints where the most likely solution is roughly mid-way between the minimum and maximum.

If the flute is most likely about 350 mm long, try constraining the bore length between 300 and 400 mm, or even tighter bounds.   You probably want T1 to be not much higher than halfway up the bore, so the total distance between T1 and the end of the bore should be not more than, say, 200 mm.  That suggests the constraint maximums for the hole separations should be more like 30, 30, 40, 30, 30, 40.  You could maybe increase the last maximum a bit, at least for use with a non-global optimizer.

NetOneGames, 2023-05-26:
Ok that is me being totally dumb then, need to be much more thorough with constraints, also really appreciate the extra information.  That has made a few things much clearer.  Thanks for taking the time.

closed 2023-05-26 by NetOneGames