edwardkort/WWIDesigner #156: Shortcuts

twentyfortysix, 2023-12-08:
Hi

I have many complains about the interface actually that I'm not gonna trow on you as I know it will be overwhelming, but I have one proposal which I feel would not be hard to implement and will help us all be more productive.
And that is shortcuts. I know there are some kind of shortcuts, but they are not really short.

Like for example to Optimize instrument I need do pres [alt]+[t] to open the tool menu and then [alt]+[o] to run the optimization.
Same things applies to all common functions one would like to use, show graph, show calculate tuning etc.

If there will be a set of simple shortcuts it would be much faster and easier to use the program. As In my case for example i do lot of manual adjustments and the need to grab a mouse and click and adjust and click and.. and quite tiresome.

note:
It would also be nice if the tuning open opens/updates after every optimization automatically as that is naturally something one would like to see and it is simply again tiresome to open the "calculate tuning" window manually each time. I know some people might complain that they do want see say graph, well that can be all part of the user settings.

(I see a lot of improvements there can be made if anybody is open to hear them.)

edwardkort, 2023-12-10:
Hi @twentyfortysix,

WIDesigner is open source. Your code contributions are welcome.

Cheers,
@edwardkort

twentyfortysix, 2023-12-10:
Hello Edward,
Sure ;)
but java has never been my thing, more over the UI obsolete library used ..
The point is that the Acoustic engine behind the program is well done and it is pity that the gui kills it.
If it would be written in https://openwind.inria.fr/ and maybe KIVY GUI it might be interesting journey but unfortunately java.

What I can offer in this case is to do the UI enhancement summary and if anybody is interested to dig in to it they have a list what to do. It is not just UI, also some useful features, like partial optimization and flute segmentation and so on.

For now I'm slowly getting used to the quirks and getting my way around.
I try to build a single reed instruments and the description how to use the program is quite short in that regard. Especially how to implement the mouth piece which is so crucial for the rest of the calculation. So far I found a workaround in a fake mouth piece shape that somehow - hopefully will work. But it is a dirty hack, It never gave me full chromatic instrument yet.. the tones are always shifted in it's own step-wise way and different from expectation/tuning. Maybe one day... 

It might not look like but I'm glad you made the program.
sincerely 2046